Xilinx/Vitis-AI #916: can not generate deploy_model when quantize_eval_model in Vitis-ai 2.5

kayky233, 2022-07-13:
ENV setup:
I used  `docker pull xilinx/vitis-ai-cpu:latest  ` and `./docker_run.sh xilinx/vitis-ai-cpu:latest` to download and use the Vitis-ai 2.5.
and followed the Introduction in /setup/mpsoc to Setup cross-compiler.
But when I try the tutorial in Vitis-AI-Tutorials/Feature_Tutorials/03-edge-to-cloud)/files,I can not generate the deploy_model .

here is the quantize info output.
2022-07-12 19:15:06.698226: W tensorflow/contrib/decent_q/utils/quantize_utils.cc:841] [DECENT_WARNING] Convert alpha of leaky_relu node leaky_re_lu_72/LeakyRelu from 0.1 to 26/256 to simulate DPU.
INFO: Calibrating for 200 iterations...
100% (200 of 200) |######################| Elapsed Time: 0:17:25 Time:  0:17:25
INFO: Calibration Done.
INFO: skip create deploy_model.pb
********************* Quantization Summary *********************      
INFO: Output:       
  quantize_eval_model: vai_q_output/quantize_eval_model.pb 

I can not find the source code of vai_q_tensorflow and i can not understand why it's skipping create deploy_model.pb.

anirudhitagi, 2022-07-13:
This is not a solution, but vai_c_tensorflow will succesfully compile with the quantize_eval_model.pb and generate xmodel. 
Not sure if it will generate desired results ( I haven't tested the xmodel on the board).

Could it be that deploy_model.pb functionalities are preset in quantize_eval_model.pb?

Thanks and Regards,
-AI

kayky233, 2022-07-13:
> This is not a solution, but vai_c_tensorflow will succesfully compile with the quantize_eval_model.pb and generate xmodel. Not sure if it will generate desired results ( I haven't tested the xmodel [...]

yeah,for most platform , quantize_eval_model.pb is enough. But for Zynq DPUCZDX8G ( which I am used ), deploy_model.pb is required.
reference is here:
in tutorial in Vitis-AI-Tutorials/Feature_Tutorials/03-edge-to-cloud)
The Vitis AI compiler will convert, and optimize where possible, the quantized model to a set of micro-instructions and then output them to either .elf or .xmodel files depending on the value of the --arch argument passed to the vai_c_tensorflow command.

Note that the input graph to the two compile commands is different. For the Zynq DPUCZDX8G it is deploy_model.pb and for the Alveo DPUCAHX8H it is quantize_eval_model.pb.

anirudhitagi, 2022-07-14:
Hi @qianglin-xlnx and @davidxudong,
Any advice or inputs with respect to this?

Thanks & Regards,
-AI

xieyi4650, 2022-07-21:
Hi @kayky233   @anirudhitagi 

deploy_model.pb is deprecated, using quantize_eval_model.pb instead
The document is not updated in time

anirudhitagi, 2022-07-21:
Hi @xieyi4650 ,
Thank you for your response!
A follow up question regarding that, the deploy_model.pb is deprecated from which version of the Vitis-AI? Is it perhaps 1.3.411 onwards?
(My reason for asking is I am experimenting with PYNQ-DPU overlay and would like some clarity on the flow)

xieyi4650, 2022-07-22:
yes, 
1.3 has deploy_model.pb
1.4 deprecated deploy_model.pb

kayky233, 2022-07-22:
> yes, 1.3 has deploy_model.pb 1.4 deprecated deploy_model.pb

yeah,thanks.I go back to use 1.3 and got the deploy_model.pb.But is it enough just using quantize_eval_model.pb?,I didn't test yet.By the way,the document is soooo confusing.

xieyi4650, 2022-07-25:
@kayky233 
quantize_eval_model.pb is enough
we will update the document.

kayky233, 2022-07-25:
> 

thank you very much.

closed 2022-09-06 by renjiangyong