Xilinx/Vitis-AI #920: ofa_yolo：no training script for float model

Gang-Chen-China, 2022-07-13:
Hi，when I download the ofa_yolo project (https://www.xilinx.com/bin/public/openDownload?filename=pt_OFA-yolo_coco_640_640_48.88G_2.5.zip)，I find there is no script to train the model.  

[image]

wangxd-xlnx, 2022-07-18:
Hi @Gang-Chen-China 

This ofa_yolo model is a typical OFA model. It's directly extracted from a supernet and inherits the weights of supernet. If we want to re-train it, in fact we need to re-train supernet. 
OFA is short for 'Once-for-All', for its more details you could refer to https://ofa.mit.edu/.

Gang-Chen-China, 2022-07-19:
> Hi @Gang-Chen-China
> 
> This ofa_yolo model is a typical OFA model. It's directly extracted from a supernet and inherits the weights of supernet. If we want to re-train, in fact we need to re-train supernet. So we did not provide train scripts for these OFA models. OFA is short for 'Once-for-All', for its more details you could refer to https://ofa.mit.edu/.

Thank you. I will try to look the paper.

Gang-Chen-China, 2022-07-19:
> Hi @Gang-Chen-China
>
> This ofa_yolo model is a typical OFA model. It's directly extracted from a supernet and inherits the weights of supernet. If we want to re-train, in fact we need to re-train [...]

Thank you, I get it. Later I will look the paper.

Gang-Chen-China, 2022-07-20:
> Hi @Gang-Chen-China
>
> This ofa_yolo model is a typical OFA model. It's directly extracted from a supernet and inherits the weights of supernet. If we want to re-train, in fact we need to re-train [...]

Hi @wangxd-xlnx，I have seen the OFA net. And I want to  know what's the difference between the ofa_yolo and yolov5 https://github.com/ultralytics/yolov5

mxjecho, 2022-11-16:
Hi @Gang-Chen-China, Ofa-yolo models are searched from YOLOV5-m. The search space only contains the output channel number; the largest model is the original YOLOV5-m.

qianglin-xlnx, 2023-01-17:
Closing since no activity for more than 2 months, please reopen if you still have any questions, thanks

closed 2023-01-17 by qianglin-xlnx