Xilinx/Vitis-AI #923: Quantization of Object Detection from Torchvision failing

lino-alves, 2022-07-14:
Hi,

I have been trying to quantize a couple of object detection models from Torchvision (Faster R-CNN and SSDlite) without success.
Currently I am using Vitis-AI 2.5. Both NNs fail with the same error.
I did a bit of debugging and I found that some of the TorchNodes in the nodes_list of the slice_to_strided_slice function in opt_pass.py are "lost" (to the lack of a better word). Interestingly this error was not happening in Vitis-AI 2.0, the error was another one already during the processing of the ops, so a later stage of the quantization.

Pasting an image of the error to keep the formatting:
[image]

Is this something I can fix or configure the quantizer to bypass?
I understand that the object detection models are not explicitly mentioned in the supported models of the quantizer, but I suppose I can customize my model to fit into the limitations of the quantizer. I am just wondering if this is a bug in the quantizer that can be fixed before I move into tweaking my models.

Final question regarding the verbosity of the quantizer. How do I increase the verbosity? I saw there is a NndctOption.nndct_parse_debug, is this the way to go or is there something else? What are the exact lines of code to change this parameter?

carrotandjzy, 2022-07-28:
vitis-ai 2.5 does not support Faster R-CNN and SSDlite. It is not easy to configure a quantizer for these cases without any modification. If you try to customize these models, you'd be better to quantize only the backbone part of the neural network and moving the rest out of the nn.Module's forward function

lino-alves, 2022-07-28:
Thanks for the reply.
This is the direction I headed. Essentially trim a good portion of the operations out of the forward function. (The backbone and head are still inside.)
Still work in progress, but I think I am getting close.

Regarding my other question, can you help on how to increase the verbosity of the quantizer? Apparently there are some variables to do it (NndctOption.nndct_parse_debug), but I didn't figure out how to change the values on those variables.

carrotandjzy, 2022-08-02:
You can append "--nndct_parse_debug=1" on  your command line to enable debug info for parser. For example,  "python xxx.py --nndct_parse_debug=1"

carrotandjzy, 2022-08-02:
A log named "nndct_debug.log" will be generated in the execute directory.

carrotandjzy, 2022-08-02:
Most limitations come from the compiler and hardware, not from the framework.

closed 2022-08-11 by niuxjxlnx