Xilinx/Vitis-AI #924: Pytorch FADNet rebuild xmodel can't execute kv260 1.4 PetaLinux dpu_task fadnet

zhihaoTW, 2022-07-17:
Hi
I trying to re-build the FADNet model, so I download Vitis AI 1.4 pt_fadnet_sceneflow_576_960_359G_1.4.zip,
then compile inside the quantization folder the FADNet_0_int, FADNet_1_int, and FADNet_2_int to xmodel, but it can't run kv260 Petalinux 1.4 it showed the following error:
[image]
I compare the Vitis AI MODEL ZOO 1.4 three FADNet xmodel, occur the FADNet_0 structure problem, it re-build xmodel is different output_tensor, my xmodel output_tensor is 1, the MODEL ZOO is 3.
my xmodel:
[image]
[image]
MODEL-ZOO:
[image]
[image]
How can I increase output_tensor? 
like the Vitis ai zoo xmodel structure result.

using compile.sh file ->
source compile.sh kv260 ../quantization/FADNet_0_int.xmodel FADNet_0_pt ./kv260/FADNet_0
[image]
[image]

max271828, 2022-07-18:
Hi,
I found your command line need change when on board terminal. 
./demo_fadnet demo_fadnet_left.jpg demo_fadnet_right.jpg
The 1.4 version demo_fadnet need to execute like this.

zhihaoTW, 2022-07-18:
> Hi, I found your command line need change when on board terminal. ./demo_fadnet demo_fadnet_left.jpg demo_fadnet_right.jpg The 1.4 version demo_fadnet need to execute like this.

Thanks for your advice.
I follow README:
[image]
and then I have replace FADNet_0 to MODELZOO xmodel, it can work .
[image]

closed 2022-07-27 by qianglin-xlnx