Xilinx/Vitis-AI #929: Vitis 2.5 Run on ZCU104 Errors

aimilefth, 2022-07-20:
I created a ResNet50 .xmodel using Vitis AI 2.5 compiled for ZCU104 and tried to test it on the ZCU104. The ZCU104 has the Vitis 2.5 image that was given on the userguide. 

The .xmodel that was run is this:
[customcnn.xmodel.zip](https://github.com/Xilinx/Vitis-AI/files/9149624/customcnn.xmodel.zip)

Running "xdputil benchmark ./customcnn.xmodel 1 -i 1" I got this error
<p>WARNING: Logging before InitGoogleLogging() is written to STDERR<br />W0720 12:21:38.937422&nbsp;&nbsp; 948 xrt_cu.cpp:188] cu timeout! device_core_idx 0&nbsp; handle=0xffff880228d0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010&nbsp; bo=1 is_done 0 <br />I0720 12:21:38.937528&nbsp;&nbsp; 948 xrt_cu.cpp:99] Total: 10010217us&nbsp;&nbsp; &nbsp;ToDriver: 18446742595559328us&nbsp;&nbsp; &nbsp;ToCU: 0us&nbsp;&nbsp; &nbsp;Complete: 0us&nbsp;&nbsp; &nbsp;Done: 1488160440us<br />F0720 12:21:38.937566&nbsp;&nbsp; 948 dpu_control_xrt_edge.cpp:186] dpu timeout! core_idx = 0<br />&nbsp;LSTART 19&nbsp; LEND 15&nbsp; CSTART 1&nbsp; CEND 1&nbsp; SSTART 0&nbsp; SEND 0&nbsp; MSTART 1&nbsp; MEND 1&nbsp; CYCLE_L 3003115961&nbsp; CYCLE_H 0 <br />*** Check failure stack trace: ***<br />/usr/bin/xdputil: line 20:&nbsp;&nbsp; 947 Aborted&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; /usr/bin/python3 -m xdputil $*<br /><br /></p>

xdputil query 
<p>{<br />&nbsp;&nbsp;&nbsp; "DPU IP Spec":{<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "DPU Core Count":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "IP version":"v4.0.0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "generation timestamp":"2022-05-14 14-30-00",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "git commit id":"4772d51",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "git commit time":2022051323,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "regmap":"1to1 version"<br />&nbsp;&nbsp;&nbsp; },<br />&nbsp;&nbsp;&nbsp; "VAI Version":{<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "libvart-runner.so":"Xilinx vart-runner Version: 2.5.0-27b0b58ebad27e3add589ee4238e05ce2aead29a&nbsp; 2022-06-08-05:41:17 ",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 2.5.0-3b292b7ea57f197f99757826bef0a3d022822630&nbsp; 2022-05-31 03:35:47 [UTC] ",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "libxir.so":"Xilinx xir Version: xir-7387c9cc7f9ad438b09006f11854ac7bab176a86 2022-06-08-05:39:16",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "target_factory":"target-factory.2.5.0 9e04a0e4c5f6c6252d7d269813cdb70a67b79b16"<br />&nbsp;&nbsp;&nbsp; },<br />&nbsp;&nbsp;&nbsp; "kernels":[<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; {<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "DPU Arch":"DPUCZDX8G_ISA1_B4096",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "DPU Frequency (MHz)":300,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "IP Type":"DPU",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load Parallel":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load augmentation":"enable",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load minus mean":"disable",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Save Parallel":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "XRT Frequency (MHz)":300,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_addr":"0x80000000",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_handle":"0xaaaae5ea55b0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_idx":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_mask":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_name":"DPUCZDX8G:DPUCZDX8G_1",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "device_id":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "fingerprint":"0x101000016010407",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "name":"DPU Core 0"<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; },<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; {<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "DPU Arch":"DPUCZDX8G_ISA1_B4096",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "DPU Frequency (MHz)":300,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "IP Type":"DPU",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load Parallel":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load augmentation":"enable",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Load minus mean":"disable",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "Save Parallel":2,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "XRT Frequency (MHz)":300,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_addr":"0x80001000",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_handle":"0xaaaae5ea5a60",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_idx":1,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_mask":1,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "cu_name":"DPUCZDX8G:DPUCZDX8G_2",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "device_id":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "fingerprint":"0x101000016010407",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "name":"DPU Core 1"<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; }<br />&nbsp;&nbsp;&nbsp; ]<br />}<br /><br /><br /><br /></p>

xdputil status before the run:
<p>{<br />&nbsp;&nbsp;&nbsp; "kernels":[<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; {<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "addrs_registers":{<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_0":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_1":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_2":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_3":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_4":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_5":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_6":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "dpu0_base_addr_7":"0x0"<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; },<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "common_registers":{<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "ADDR_CODE":"0x0",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "AP status":"idle",<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "CONV END":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "CONV START":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "HP_ARCOUNT_MAX":7,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "HP_ARLEN":15,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "HP_AWCOUNT_MAX":7,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "HP_AWLEN":15,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "LOAD END":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "LOAD [... cut ...] bsp;&nbsp;&nbsp;&nbsp;&nbsp; "SAVE END":0,<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "SAVE START":0<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; },<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "name":"DPU Registers Core 1"<br />&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; }<br />&nbsp;&nbsp;&nbsp; ]<br />}<br /><br /></p>

Let me know if I can provide more information on the problem.

Thanks

xihuazhanghui, 2022-07-21:
Hi， aimilefth

We tried to reproduce the issues, but we run successfully. Please check if your environment or code is clean. Or try the models we provide in the default zcu104 images.
xdputil status before the run(same with issue provided ): 

root@xilinx-zcu104-20221:~# xdputil status
{
    "kernels":[
        {
            "addrs_registers":{
                "dpu0_base_addr_0":"0x0",
                "dpu0_base_addr_1":"0x0",
                "dpu0_base_addr_2":"0x0",
                "dpu0_base_addr_3":"0x0",
                "dpu0_base_addr_4":"0x0",
                "dpu0_base_addr_5":"0x0",
                "dpu0_base_addr_6":"0x0",
                "dpu0_base_addr_7":"0x0"
            },
            "common_registers":{
                "ADDR_CODE":"0x0",
                "AP status":"idle",
                "CONV END":0,
                "CONV START":0,
                "HP_ARCOUNT_MAX":7,
                "HP_ARLEN":15,
                "HP_AWCOUNT_MAX":7,
                "HP_AWLEN":15,
                "LOAD END":0,
                "LOAD START":0,
                "MISC END":0,
                "MISC START":0,
                "SAVE END":0,
                "SAVE START":0
            },
            "name":"DPU Registers Core 0"
        },
        {
            "addrs_registers":{
                "dpu0_base_addr_0":"0x0",
                "dpu0_base_addr_1":"0x0",
                "dpu0_base_addr_2":"0x0",
                "dpu0_base_addr_3":"0x0",
                "dpu0_base_addr_4":"0x0",
                "dpu0_base_addr_5":"0x0",
                "dpu0_base_addr_6":"0x0",
                "dpu0_base_addr_7":"0x0"
            },
            "common_registers":{
                "ADDR_CODE":"0x0",
                "AP status":"idle",
                "CONV END":0,
                "CONV START":0,
                "HP_ARCOUNT_MAX":7,
                "HP_ARLEN":15,
                "HP_AWCOUNT_MAX":7,
                "HP_AWLEN":15,
                "LOAD END":0,
                "LOAD START":0,
                "MISC END":0,
                "MISC START":0,
                "SAVE END":0,
                "SAVE START":0
            },
            "name":"DPU Registers Core 1"
        }
    ]
}
xdputil query( same with provided):

root@xilinx-zcu104-20221:~# xdputil query
{
    "DPU IP Spec":{
        "DPU Core Count":2,
        "IP version":"v4.0.0",
        "generation timestamp":"2022-05-14 14-30-00",
        "git commit id":"4772d51",
        "git commit time":2022051323,
        "regmap":"1to1 version"
    },
    "VAI Version":{
        "[libvart-runner.so](http://libvart-runner.so/)":"Xilinx vart-runner Version: 2.5.0-27b0b58ebad27e3add589ee4238e05ce2aead29a  2022-06-08-05:41:17 ",
        "[libvitis_ai_library-dpu_task.so](http://libvitis_ai_library-dpu_task.so/)":"Xilinx vitis_ai_library dpu_task Version: 2.5.0-3b292b7ea57f197f99757826bef0a3d022822630  2022-05-31 03:35:47 [UTC] ",
        "[libxir.so](http://libxir.so/)":"Xilinx xir Version: xir-7387c9cc7f9ad438b09006f11854ac7bab176a86 2022-06-08-05:39:16",
        "target_factory":"target-factory.2.5.0 9e04a0e4c5f6c6252d7d269813cdb70a67b79b16"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B4096",
            "DPU Frequency (MHz)":300,
            "IP Type":"DPU",
            "Load Parallel":2,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":300,
            "cu_addr":"0x80000000",
            "cu_handle":"0xaaaacf700760",
            "cu_idx":0,
            "cu_mask":2,
            "cu_name":"DPUCZDX8G:DPUCZDX8G_1",
            "device_id":0,
            "fingerprint":"0x101000016010407",
            "name":"DPU Core 0"
        },
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B4096",
            "DPU Frequency (MHz)":300,
            "IP Type":"DPU",
            "Load Parallel":2,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":300,
            "cu_addr":"0x80001000",
            "cu_handle":"0xaaaacf700c10",
            "cu_idx":1,
            "cu_mask":1,
            "cu_name":"DPUCZDX8G:DPUCZDX8G_2",
            "device_id":0,
            "fingerprint":"0x101000016010407",
            "name":"DPU Core 1"
        }
    ]
}
test pass with provided .xmodel:

root@xilinx-zcu104-20221:~# xdputil benchmark ./customcnn.xmodel 1 -i 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0721 04:59:56.791045   875 test_dpu_runner_mt.cpp:475] shuffle results for batch...
I0721 04:59:56.791904   875 performance_test.hpp:73] 0% ...
I0721 05:00:02.792057   875 performance_test.hpp:76] 10% ...
I0721 05:00:08.792227   875 performance_test.hpp:76] 20% ...
I0721 05:00:14.792407   875 performance_test.hpp:76] 30% ...
I0721 05:00:20.792563   875 performance_test.hpp:76] 40% ...
I0721 05:00:26.792737   875 performance_test.hpp:76] 50% ...
I0721 05:00:32.792939   875 performance_test.hpp:76] 60% ...
I0721 05:00:38.793114   875 performance_test.hpp:76] 70% ...
I0721 05:00:44.793267   875 performance_test.hpp:76] 80% ...
I0721 05:00:50.793423   875 performance_test.hpp:76] 90% ...
I0721 05:00:56.793596   875 performance_test.hpp:76] 100% ...
I0721 05:00:56.793665   875 performance_test.hpp:79] stop and waiting for all threads terminated....
I0721 05:00:56.802091   875 performance_test.hpp:85] thread-0 processes 5817 frames
I0721 05:00:56.802124   875 performance_test.hpp:93] it takes 8433 us for shutdown
I0721 05:00:56.802145   875 performance_test.hpp:94] FPS= 96.9335 number_of_frames= 5817 time= 60.0102 seconds.
I0721 05:00:56.802201   875 performance_test.hpp:96] BYEBYE
Test PASS.

aimilefth, 2022-07-21:
Hello xihuazhanghui, thanks for the quick response.

I tried to run the test on another board and got the correct results too. I suspect that there is something wrong with the board i used. I think this issue can be considered closed.

closed 2022-07-21 by aimilefth