Xilinx/Vitis-AI #931: runtime_error  CU timeout 0

jzymessi, 2022-07-21:
When I use vck5000 run onnxruntime demo, get into error 
 
[image]

I try the flow with u200 ,it's succeed.
How can I solve this error?

SuZhan0322, 2022-07-29:
hi @jzymessi 
Which vck5000 DPU ID did you used? https://github.com/Xilinx/Vitis-AI/blob/master/setup/vck5000/README.md
Are you using vck5000 prod or vck5000 es? 
And you can first verify your machine state with the release xmodel and AILibrary in docker container.

jzymessi, 2022-08-01:
@SuZhan0322 :
     After my experiment, it seems to be compiled using VITIS-AI-2.0 environment, but running in ViTIS-AI-2.5 environment.
     By the way,I encountered two errors in feature extraction data when running the model. The feature results extracted from all pictures are the same. I have solved the problem after powering off and restarting VCK5000. May I ask how this phenomenon is caused?

SuZhan0322, 2022-08-05:
@jzymessi ：
The model's build environment version should be the same as the runtime environment version.  They need to be all VITIS-AI-2.0 or all VITIS-AI-2.5.
After the model runs timeout, the DPU state will not meet expectations. We recommend to reset the board or cold restart after DPU timeout.

closed 2022-08-17 by jzymessi