Xilinx/Vitis-AI #933: TF2 compilation ends with error without ID

ni-maxkoehler, 2022-07-22:
I tried to compile a model, which we previously deployed on VCK190, for DPUCVDX8H/VCK50004PE/arch.json.
This ends with an error output without an ID (see output below)-
Is there a way to debug the error?

```
(vitis-ai-tensorflow2) Vitis-AI /workspace/model_zoo/custom_model > vai_c_tensorflow2 -m quantized/quantized_model_dilation_rate.h5 -o compile -n custom_model -a /opt/vitis_ai/compiler/arch/DPUCVDX8H/VCK50004PE/arch.json
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized/quantized_model_dilation_rate.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/deeprx_vck5000_DPUCVDX8H_ISA1_F2W4_4PE_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/model_zoo/custom_model/quantized/quantized_model_dilation_rate.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|█████████████████████████████████| 103/103 [00:00<00:00, 29644.78it/s]
[INFO] infer shape (NHWC)  :100%|█████████████████████████████████| 176/176 [00:00<00:00, 4599.73it/s]
[INFO] perform level-0 opt :100%|█████████████████████████████████| 2/2 [00:00<00:00, 113.17it/s]
[INFO] perform level-1 opt :100%|█████████████████████████████████| 2/2 [00:00<00:00, 443.89it/s]
[INFO] generate xmodel     :100%|█████████████████████████████████| 176/176 [00:00<00:00, 5601.40it/s]
[INFO] dump xmodel: /tmp/custom_model_DPUCVDX8H_ISA1_F2W4_4PE_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8H_ISA1_F2W4_4PE
[UNILOG][INFO] Graph name: DeepRx, with op num: 420
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][ERROR_SAMPLE][Error without ID assigned occur!]
*** Check failure stack trace: ***
Aborted (core dumped)
(vitis-ai-tensorflow2) Vitis-AI /workspace/model_zoo/custom_model >
```

Vitis-AI-4836, 2022-07-29:
Hi, ni-maxkoehler,

     VCK190 and VCK5000 base on different hardware , so it has different compilation constraints. 

     This error usually happens during instruction generation.

     Could you please give me your dump xmodel file (/tmp/custom_model_DPUCVDX8H_ISA1_F2W4_4PE_org.xmodel),  so I can debug the issue  for you.

ni-maxkoehler, 2022-07-29:
Which way can I share the file so that it remains private?

Vitis-AI-4836, 2022-08-04:
> Which way can I share the file so that it remains private?
sorry for the late reply, you can mail to yunfengy@amd.com.

Vitis-AI-4836, 2022-08-04:
> Which way can I share the file so that it remains private?

sorry for the late reply, you can mail to yunfeng.yu@amd.com.

Vitis-AI-4836, 2022-08-22:
Hi, ni-maxkoehler,
VCK50004PE has a compiler hardware resource limit, which may be fixed in our next release. But you can compile your xmodel by VCK50008PE /DPUCVDX8H_ISA1_F2W2_8PE and it compiles successfully. DPUCVDX8H_ISA1_F2W2_8PE has better performance than DPUCVDX8H_ISA1_F2W4_4PE .

ni-maxkoehler, 2022-08-22:
Thank you for checking. I did try to compile for VCK50008PE already. I would have expected that the model is completely able to execute on the DPU as for the VCK190. But it seems as there is a couple of submodules which can only be executed by the CPU. Can you give an advice on what needed to get it running on the DPU?

Vitis-AI-4836, 2022-08-22:
VCK50008PE doesn't have dwconv engine, so depthwise-conv2d-fix op can only  be executed by the CPU.

ni-maxkoehler, 2022-08-22:
Is there a version of the VCK5000 which has this engine?

ni-maxkoehler, 2022-08-22:
For VCK50006PE_aieDWC i am running into the same error as originally described in this thread. Is there any workaround yet for the compiler HW resource limit?

Vitis-AI-4836, 2022-08-22:
We may support it in the next version. If you must use this VCK5000, I need to figure out a way to see if I can get you a patch.

ni-maxkoehler, 2022-08-22:
Yes we have to use VCK5000 as the VCK190 ES support has ended with the latest Vitis-AI release.

Vitis-AI-4836, 2022-09-01:
Hi ni-maxkoehler，
I'm very sorry to tell you that I don't have permission to make patches. You may have to wait until vai3.0 is released, probably at the end of 2022.

SuZhan0322, 2022-09-29:
Hi @ni-maxkoehler ,
Please reopen this issue if you have any other questions, Thanks!

closed 2022-09-29 by SuZhan0322