Xilinx/Vitis-AI #936: replace_pytorch fails

ash8327, 2022-07-22:
The Vitis AI containers ran with these issues: 
xilinx/vitis-ai-gpu:2.5
xilinx/vitis-ai-gpu:2.5.0
xilinx/vitis-ai-gpu:2.5.0.1260
xilinx/vitis-ai-gpu:latest
xilinx/vitis-ai-cpu:latest

I am trying to run the pt_OFA-yolo_coco_640_640_0.5_24.62G_2.5 readme instructions with the Vitis AI GPU container 
If I run with the initial pytorch in the container and I try to run the run_test.sh the error is:

`(vitis-ai-pytorch) Vitis-AI /workspace/model_zoo/pt_OFA-yolo_coco_640_640_0.5_24.62G_2.5/code > bash run_test.sh 
Traceback (most recent call last):
  File "test.py", line 15, in <module>
    from models.models import ofa_yolo_30,ofa_yolo_50,ofa_yolo_0
  File "/workspace/model_zoo/pt_OFA-yolo_coco_640_640_0.5_24.62G_2.5/code/models/models.py", line 17, in <module>
    from models.yolo_base_30 import Model as base_model_30
  File "/workspace/model_zoo/pt_OFA-yolo_coco_640_640_0.5_24.62G_2.5/code/models/yolo_base_30.py", line 18, in <module>
    from pytorch_nndct.nn import QuantStub, DeQuantStub
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/__init__.py", line 14, in <module>
    from .apis import *
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 25, in <module>
    from .qproc import TorchQuantProcessor
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/__init__.py", line 1, in <module>
    from .base import TorchQuantProcessor, dump_xmodel
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 30, in <module>
    from pytorch_nndct.quantization import TORCHQuantizer, FakeQuantizer
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/quantization/__init__.py", line 2, in <module>
    from .torch_qalgo import *
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/quantization/torch_qalgo.py", line 28, in <module>
    from pytorch_nndct.nn import fake_quantize_per_tensor, fake_quantize_per_channel
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/__init__.py", line 1, in <module>
    from pytorch_nndct.nn.modules import functional
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/__init__.py", line 16, in <module>
    from .sigmoid import *
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/sigmoid.py", line 26, in <module>
    from .fix_ops import NndctSigmoidTableLookup, NndctSigmoidSimulation
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/fix_ops.py", line 26, in <module>
    from ..load_kernels import *
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/load_kernels.py", line 31, in <module>
    torch.ops.load_library(lib_abspath)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/torch/_ops.py", line 104, in load_library
    ctypes.CDLL(path)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/_kernels.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZNK2at10TensorBase8toStringEv
`
Then I try to replace pytorch using the script and it fails with:
`
Looking for: ['pytorch==1.9.0', 'torchvision==0.10.0', 'torchaudio==0.9.0', 'cudatoolkit=10.2']

pytorch/noarch           [====================] (00m:00s) Done
pytorch/linux-64         [====================] (00m:00s) Done
pkgs/main/noarch         [====================] (00m:00s) Done
pkgs/r/linux-64          [====================] (00m:00s) Done
pkgs/r/noarch            [====================] (00m:00s) Done
pkgs/main/linux-64       [====================] (00m:00s) Done

Pinned packages:
  - python 3.7.*

Encountered problems while solving:
  - nothing provides cudatoolkit >=11.1,<11.2 needed by torchvision-0.10.0-py36_cu111

#### Pytorch packages is NOT replaced correctly.
`
If I try to replace pytorch with another version in the script it also doesn't work:
#### Pytorch packages is NOT replaced correctly.

niuxjxlnx, 2022-08-11:
@ash8327 :

Please use the updated script at https://github.com/Xilinx/Vitis-AI/blob/master/docker/dockerfiles/replace_pytorch.sh.

closed 2022-08-31 by niuxjxlnx

anirudhitagi, 2023-02-13:
Greetings,

I get a `CondaHTTPError: HTTP 000 CONNECTION FAILED` message when I use the replace_pytorch.sh script. I have described the error here - https://github.com/Xilinx/Vitis-AI/issues/1170 

@niuxjxlnx and @ash8327 have you faced a similar error before? 
Please advise how to resolve this?