Xilinx/Vitis-AI #937: vai_c_tensorflow error: IndexError: list index out of range

josepedro27, 2022-07-22:
Hi all, 
I'm following the example: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/07-yolov4-tutorial
I'm trying to compile the yolov5 quantized model with the options:

[image]

But i get the following error:

[image]

any suggestion? Thanks 

mega link with quantized and pb model - https://mega.nz/folder/bssiRRxZ#AZx711omDtOlv9pxVoKqlg

zhenzhen-AMD, 2022-08-03:
Hi @josepedro27 ,
We're working to resolve this issue, and we'll let you know if there's progress.

zhenzhen-AMD, 2022-09-06:
Hi @josepedro27 ,
Can you provide floating point model code?

qianglin-xlnx, 2023-04-27:
Closing since no activity for more than half year, please reopen if you still have question, thanks

closed 2023-04-27 by qianglin-xlnx