Xilinx/Vitis-AI #938: How to train ofa_yolo?

Gang-Chen-China, 2022-07-23:
Hi, thanks for your update. 

I want to know how do you train once-for-all model for yolo? Can you give some tutorials? 

There is pretrained ofa_yolo model in the model zoo. How do you select the sub-networks?

wangxd-xlnx, 2022-07-28:
Hi @Gang-Chen-China 

At first, user need train a full model which is the largest model here. Then with designed search space and progressive shrinking you will get many sub-nets. 

While select the specified sub-network, the searching process is under constraints that including latency and accuracy. User need to set latency predictor and accuracy predictor to find the specialized sub-nets.

For more details and tutorial, you could refer to https://github.com/mit-han-lab/once-for-all.

qianglin-xlnx, 2022-08-31:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-08-31 by qianglin-xlnx