Xilinx/Vitis-AI #939: vai_utf issue

LorenzoSun-V, 2022-07-25:
When I use inspector in vai_q_pytorch, I used target="DPUCVDX8H_ISA1_F2W2_8PE"(DPU IP of VCK5000) to init pytorch_nndct.Inspector but got the following error:
[image]

My code is like:
[image]

When I use other DPU IP like "DPUCZDX8G_ISA1_B512", it can run correctly.

carrotandjzy, 2022-08-17:
The reason is that the proto in target factory is not the same with that in  vai_utf.

LorenzoSun-V, 2022-08-17:
Thanks, I'll double check the target name.

ye33322, 2022-09-05:
Hello xilinx,

When I used target="DPUCVDX8H_ISA1_F2W4_6PE_aieDWC" the above problem also occurs. For vck5000, what target should be used?

carrotandjzy, 2022-10-08:
I think any target with "load_mode" field will failed in vai-2.5. You can skip inspect step to do quantization.