Xilinx/Vitis-AI #940: RuntimeError: Unsupported op with parameters: upsample_1(generic)

acg93-pixel, 2022-07-25:
Hi, I'm working with vitis-ai-optimizer_tensorflow2 environment in Vitis AI 2.5. Following the "mnist_convnet" example with my stacked hourglass alike model, I'm getting an error:

yuwang-xlnx, 2022-07-28:
Hi @acg93-pixel,

The supported operations in VAI Optimizer 2.5: 

- layers.Conv2D
- layers.DepthwiseConv2D
- layers.Dense
- layers.BatchNormalization

Conv2DTranspose is not supported currently and will be supported in Vitis 3.0

acg93-pixel, 2022-07-28:
Thanks, when is Vitis 3.0 release date?

yuwang-xlnx, 2022-07-28:
> Thanks, when is Vitis 3.0 release date?

About January next year.

Qiang-Feng98, 2023-08-06:
@yuwang-xlnx 
Hi, When I prune my model, I meet an error: **_nndct_shared.pruning.errors.OptimizerNotExcludeNodeError: Must exclude node from pruning: FastCS::FastCS/Sequential[decoder]/ConvTranspose2d[0]/ret.7._**
I tried to solve this with the **_excludes_** parameter: **_excludes = ['FastCS::FastCS/Sequential[decoder]/ConvTranspose2d[0]/ret.7' ], pruing_runner.ana(eval_fn, gpus, excludes=excludes)_** 
but it didn't work. How should I use the **_excludes_** parameter?
I used Vitis-AI 3.0 on Ubuntu 20.04. Thanks.