Xilinx/Vitis-AI #950: Quantization of a PyTorch model stalls during evaluation loop

daniperfer, 2022-08-03:
Hi,

I'm trying to quantize a pytorch model with Vitis-AI. I have a quantization script which first calls to `torch_quantizer`, and then do an evaluation loop which makes inference on the "quantized" model. However, the execution stalls at this point, during the evaluation loop.
Do anybody have any thoughts on what could be going wrong for having this behavior? 

The model is a slightly modified version of VGG19, which allows the extraction of some intermediate layers:
```
class Vgg19(torch.nn.Module):
    # modified from the original at
    # https://github.com/chenyuntc/pytorch-book/blob/master/chapter08-neural_style/PackedVGG.py

    def __init__(self, batch_normalization=True, required_layers=None):

        if required_layers == None and batch_normalization:
            required_layers = [3, 10, 17, 30, 43, 46]
        elif required_layers == None:
            required_layers = [2, 7, 12, 21, 30, 32]

        # features 2，7，12，21, 30, 32: conv1_2,conv2_2,relu3_2,relu4_2,conv5_2,conv5_3
        super(Vgg19, self).__init__()

        if batch_normalization:
            features = list(models.vgg19_bn(pretrained=True).features)[
                :47]  # get vgg features
            self.avgpool = models.vgg19_bn(pretrained=True).avgpool
            self.classifier = models.vgg19_bn(pretrained=True).classifier
        else:
            features = list(models.vgg19(pretrained=True).features)[
                :33]  # get vgg features
            self.avgpool = models.vgg19(pretrained=True).avgpool
            self.classifier = models.vgg19(pretrained=True).classifier

        # construct network in eval mode
        self.features = torch.nn.ModuleList(features).eval()

        for param in self.features.parameters():  # we don't need gradients, turn them of to save memory
            param.requires_grad = False

        # record required layers to save them in forward
        self.required_layers = required_layers

        for layer in required_layers[:-1]:
            # do not overwrite in any layer
            self.features[layer+1].inplace = False

    def forward(self, x):
        results = []
        for ii, model in enumerate(self.features):
            x = model(x)
            if ii in self.required_layers:
                results.append(x)
        # pass x to head and get output
        x = self.avgpool(x)
        flatten = torch.flatten(x, 1)
        output = self.classifier(flatten)

        return results, output
```

The quantization script is as follows:
```
def load_imagenet_data(
        data_dir='../../data/imagenet-mini',
        batch_size=128,
        subset_len=None,
        sample_method='random',
        target_shape = (224, 224),
        **kwargs):
    # prepare data
    valdir = data_dir + '/val'
[... 232 lines cut ...]
[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing Vgg19...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 44/44 [00:00<00:00, 2050.00it/s, OpInfo: name = return_0, type = Return]

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(./quantization_results_vgg19/Vgg19.py)

[VAIQ_NOTE]: =>Get module with quantization.
2022-08-03 09:54:43.543 | INFO     | __main__:quantize:419 - Evaluation Quantized model
2022-08-03 09:54:43.545 | DEBUG    | __main__:evaluate_imagenet:167 - len(val_loader) 100
  0%|                                                                                                                                                                      | 0/100 [00:00<?, ?it/s]> /opt/vitis_ai/
---- EXECUTINON STALLS HERE ----
```

Do anybody have any thoughts on what could be going wrong for having this behavior?

niuxjxlnx, 2022-08-11:
@daniperfer :
 
Can you paste the python stack here? You can type CTRL+C or use pdb to get the stack.
Other information may helps (type nvidia-smi to get the information):
1) GPU dirver version and GPU card architecture in your server
2) How many memory on the GPU card

daniperfer, 2022-08-29:
Thanks for replying, @niuxjxlnx:

I've tried to type CTRL+C when the execution of my script stalled, but it hasn't respond to the command. So I've used pdb for running it step by step. I've run it several times so as to find what was the last line which I could run. Finally, I think it occurs just after the line 
```
"/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/fix_ops.py", line 132, in NndctDiffsFixPos
    torch.ops.vai.diffs_fix_pos(Tinput, Tbuffer, Tfixpos, bit_width, range, method, device_id)
```

The whole stack is as follows:
```
[VAIQ_NOTE]: Loading NNDCT kernels...                                                                                                                                                                              

-----------------------------------------                                                                                                                                                                          
PyTorch version :  1.10.1                                                                                                                                                                                          
3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53)                                                                                                                                                
[GCC 9.4.0]                                                                                                                                                                                                        
-----------------------------------------                                                                                                                                                                          
 Command line options:                                                                                                                                                                                             
--dset_dir     :  ../../data/imagenet-mini                                                                                                                                                                         
--ann_file     :  dataset/annotation/image_info_test2017.json                                      
--float_model  :  VGG19                                                              
--quant_mode   :  calib                                                                                                     
--batch_size   :  2                                                                                                 
--quant_output :  ./quantization_results_vgg19                                                               
--subset_len   :  200                                                                  
--deploy       :  False                                                                                                      
--input_pairs  :  image_pairs_jp.txt                                         
-----------------------------------------

2022-08-29 14:22:03.531 | INFO     | __main__:quantize:283 - You have 4 CUDA devices available
2022-08-29 14:22:03.588 | INFO     | __main__:quantize:293 - Selecting device with less allocated memory...
2022-08-29 14:22:03.589 | DEBUG    | __main__:quantize:294 - Devices memory status (sorted) [('GeForce RTX 3090', 24268.0, 2107.0), ('GeForce RTX 3090', 24268.0, 2.0), ('GeForce RTX 2080 Ti', 11019.0, 3.0), ('GeForce RTX 2080 Ti', 11019.0, 3.0)]                                                        
2022-08-29 14:22:03.589 | INFO     | __main__:quantize:308 - loading VGG19...                                 
2022-08-29 14:22:13.102 | INFO     | __main__:quantize:310 - model is loaded.                                                                                                                                      
2022-08-29 14:22:14.445 | INFO     | __main__:quantize:329 - load detector from: VGG19                            
2022-08-29 14:22:14.446 | INFO     | __main__:quantize:330 - device = cuda:1

2022-08-29 14:22:15.463 | INFO     | __main__:quantize:363 - evaluate_imagenet Float model
2022-08-29 14:22:15.464 | DEBUG    | __main__:evaluate_imagenet:167 - len(val_loader) 100
loss: 3.45729
top-1 / top-5 accuracy: 0.5 / 0.5
2022-08-29 14:22:21.333 | INFO     | __main__:quantize:386 - Quantization
quantize_vgg19_features_vai25.py:390: DeprecationWarning: This function is deprecated. Please call randint(0, 255 + 1) instead
  0, 255, target_shape + (3,)).astype(np.uint8)

[VAIQ_NOTE]: Quant config file is empty, use default quant configuration

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing Vgg19...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 44/44 [00:00<00:00, 2044.82it/s, OpInfo: name = return_0, type = Return]

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(./quantization_results_vgg19/Vgg19.py)

[VAIQ_NOTE]: =>Get module with quantization.

2022-08-29 14:22:24.949 | INFO     | __main__:quantize:418 - Evaluation Quantized model
> /workspace/andante/MM-PR-01326-ANDANTE-image_registration/quantize_vgg19_features_vai25.py(424)quantize()
[... cut ...]
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      8504      C   /opt/conda/bin/python            2105MiB |
|    0   N/A  N/A     16402      C   python                           1699MiB |
|    2   N/A  N/A     16402      C   python                           7947MiB |
+-----------------------------------------------------------------------------+
```

I am using the Vitis-AI docker version 2.5.

niuxjxlnx, 2022-08-31:
@daniperfer 

How about the status of "--quant_mode float"? I mean inference the model without quantization. Stalls during inference or not?

niuxjxlnx, 2022-09-01:
@daniperfer 
In the following part of code, what's the shape of rand_in? 
```
    if quant_mode == 'float':
        quantized_model = model
    else:
        logger.info(f"Quantization")
        rand_in = []
        for _ in np.arange(batch_size):
            input_sample = np.random.random_integers(
                0, 255, target_shape + (3,)).astype(np.uint8)
            # just normalize input image and add padding to have dimensions multiple of 16
            rand_tensor, _ = dfm_transform(
                input_sample, device, padding_n=padding_n, mean=image_mean, std=image_std)
            rand_in.append(rand_tensor)

        inputs = torch.cat(rand_in, dim=0)
        quantizer = torch_quantizer(
            quant_mode, model, (inputs), device=device, output_dir=quant_output)
```

daniperfer, 2022-09-01:
Hi, @niuxjxlnx:

`rand_in` is a list of length B (B for batch length) which contains three-dimesional pytorch tensors (which are supposed to be images).

I've tested it with batches 1, 2 and 4, and I've set width=height for the rand images; and I've got the following shapes for the tensors inside `rand_in`:

with batch 1:
```
2022-09-01 12:20:37.537 | DEBUG    | __main__:quantize:396 - len(rand_in)=1
2022-09-01 12:20:37.537 | DEBUG    | __main__:quantize:398 - rand_in[0].shape=torch.Size([1, 3, 640, 640])
2022-09-01 12:20:37.538 | DEBUG    | __main__:quantize:406 - inputs.shape = torch.Size([1, 3, 640, 640])
```

with batch 2:
```
2022-09-01 12:22:27.610 | DEBUG    | __main__:quantize:396 - len(rand_in)=2
2022-09-01 12:22:27.611 | DEBUG    | __main__:quantize:398 - rand_in[0].shape=torch.Size([1, 3, 640, 640])
2022-09-01 12:22:27.612 | DEBUG    | __main__:quantize:398 - rand_in[1].shape=torch.Size([1, 3, 640, 640])
2022-09-01 12:22:27.613 | DEBUG    | __main__:quantize:406 - inputs.shape = torch.Size([2, 3, 640, 640])
```

with batch 4:
```
2022-09-01 15:49:29.682 | DEBUG    | __main__:quantize:396 - len(rand_in)=4
2022-09-01 15:49:29.683 | DEBUG    | __main__:quantize:398 - rand_in[0].shape=torch.Size([1, 3, 640, 640])
2022-09-01 15:49:29.683 | DEBUG    | __main__:quantize:398 - rand_in[1].shape=torch.Size([1, 3, 640, 640])
2022-09-01 15:49:29.683 | DEBUG    | __main__:quantize:398 - rand_in[2].shape=torch.Size([1, 3, 640, 640])
2022-09-01 15:49:29.683 | DEBUG    | __main__:quantize:398 - rand_in[3].shape=torch.Size([1, 3, 640, 640])
2022-09-01 15:49:29.684 | DEBUG    | __main__:quantize:406 - inputs.shape = torch.Size([4, 3, 640, 640])
```

daniperfer, 2022-09-02:
Hi:

I've finally found a way of avoiding the stalling problem I was having. It have to do with strictly using a single GPU for running the process.

I don't know what is exactly the reason, but the process was always running through several GPUs (I could observe that in nvidia-smi output) in spite of specifying only one of them by using the line 
`device = torch.device(f"cuda:{gpu_idx}")`
at the beginning of my script. 

So I tried to limit the number of GPUs the process could see, so as to run as if there were just a single GPU. I put the following lines at the beginning of my python script:
```
os.environ["CUDA_VISIBLE_DEVICES"] = str(gpu_idx)
device = torch.device(f"cuda:0")
```
And this way the process does not stall: the quantization script run complete until the end.

So I am using this solution from now on to prevent my process from stalling.

niuxjxlnx, 2022-09-05:
@daniperfer : 

Using different GPU cards in one server may face to extra risk of GPU memory accessing, which is also what I want you help verify. This is also the reason why I cannot repeat the same problem in my own server, on which 4 same GPU cards are installed.

closed 2022-09-05 by niuxjxlnx