Xilinx/Vitis-AI #951: vai_q_tensorflow2.x, ValueError: Cannot assign value to variable, Shape mismatch.

FrancescoTosetti, 2022-08-04:
Hi all,
I am trying to quantize Keras models with the latest version of Vitis AI (2.5). I have tried with different neural network but I always get this error
```bash
Traceback (most recent call last):
  File "/home/francesco/workspace/kria/vai_quantizer.py", line 96, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=sequence)
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 841, in quantize_model
    self.optimize_model()
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 648, in optimize_model
    self._create_optimized_model()
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 358, in _create_optimized_model
    self._optimized_model, self._layer_metadata = create_optimize_model(
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1003, in create_optimize_model
    optimized_model, layer_metadata = optimize_pipeline.apply(
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 81, in apply
    transformed_model, layer_metadata = _apply_availables(
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 52, in _apply_availables
    transformed_model, layer_metadata = model_transformer.ModelTransformer(
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 738, in recursive_transform
    return ModelTransformer(transformed_model, self.transforms,
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 706, in transform
    self._set_layer_weights(layer, weights_map)
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 595, in _set_layer_weights
    K.batch_set_value(weight_value_tuples)
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/tensorflow/python/util/traceback_utils.py", line 153, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/home/francesco/.local/share/virtualenvs/hardcode-xwU4mwSS/lib/python3.8/site-packages/keras/backend.py", line 4019, in batch_set_value
    x.assign(np.asarray(value, dtype=dtype_numpy(x)))
ValueError: Cannot assign value to variable ' Conv1/kernel:0': Shape mismatch.The variable shape (3, 3, 3, 32), and the assigned value shape (1, 1, 960, 320) are incompatible.
```

I am not sure what the problem might be, but I guess it is due to the fact that the shape of the calibration dataset is not correct.
In the official documentation, I saw that the calibration dataset can be passed as np.array or as tf.keras.utils.Sequence. I have tried both ways and get the same error. I have tried passing both the actual images of the dataset but also random data, and in both cases I get the same error.

The snippet of code to reproduce the error is the following

```python3
from cv2 import imread
import tensorflow as tf
from tensorflow.keras.utils import Sequence
import cv2
import numpy as np
from skimage.io import imread
from skimage.transform import resize
from tensorflow_model_optimization.quantization.keras import vitis_quantize, vitis_inspect
import tensorflow as tf
import glob

DEFAULT_IMAGE_SIZE = 256
NUM_CHANNELS = 3

class ImageSequence(Sequence):

    def __init__(self, x_set, batch_size, height=DEFAULT_IMAGE_SIZE, channels=NUM_CHANNELS):
        self.x = x_set
        self.batch_size = batch_size
        self.heigth = height
        self.channels = channels

    def __len__(self):
        return int(np.ceil(len(self.x) / float(self.batch_size)))

    def __getitem__(self, idx):
        batch_x = self.x[idx * self.batch_size:(idx + 1) * self.batch_size]
        item = [np.load(file_name) for file_name in batch_x] # images are stored in .npy array
        return item

DIR_PATH = './img_RGB/*.*'
images = glob.glob(DIR_PATH)
sequence = ImageSequence(images, 16)

float_model = tf.keras.models.load_model('myModel.h5')
quantizer = vitis_quantize.VitisQuantizer(float_model)

quantized_model = quantizer.quantize_model(calib_dataset=sequence) # IT FAILS HERE

quantized_model.save('quantized.h5)
```

Input tensor shape is ?,256,25,3

[image]

Any help or suggestion would be greatly appreciated!

Thank you

FrancescoTosetti, 2022-09-06:
Hi @renjiangyong, thank you for your reply. I cannot share "myModel.h5", but I can give you another model to reproduce the same issue.  [You can find it here](https://drive.google.com/file/d/1Ut9RgdipPdvTpCgU3eM-MTDEdKd-jyZW/view?usp=sharing).

The input shape now is [1,32,32,3], and the error is the following:

`ValueError: Cannot assign value to variable ' block1_conv1/kernel:0': Shape mismatch.The variable shape (3, 3, 3, 64), and the assigned value shape (3, 3, 512, 512) are incompatible.`

Thank you,

Francesco

FrancescoTosetti, 2022-10-13:
Hello @renjiangyong, any news about this?
Thank you

Francesco

FrancescoTosetti, 2022-11-16:
I don't know. If you don't mind, you could start sharing your model to see if it has any similarities with mine and figure out what the cause might be.

Thank you

alpereenkaratas, 2022-11-16:
Of course. Here is my model:
https://drive.google.com/file/d/1q_5uZUeR_7J8K6UVnK-dV68qG0fBxV0J/view?usp=share_link
Thank you @FrancescoTosetti

alpereenkaratas, 2022-11-16:
Also here #1088 I opened a git issue so you can review the details. @FrancescoTosetti

alpereenkaratas, 2022-11-16:
In this case, is the problem related to Vitis AI or is there something I can do? By the way, thanks for your interest @FrancescoTosetti.

FrancescoTosetti, 2022-11-16:
The problem seems to be related to Vitis AI. As a workaround, you could try to redefine your model without the submodels, or you can try[ flattening the model following the answer at this thread](https://stackoverflow.com/questions/54648296/how-to-flatten-a-nested-model-keras-functional-api).
For some models it's still quite complicated though, so if you don't succeed you have to wait for Xilinx 
@alpereenkaratas 

Francesco

alpereenkaratas, 2022-11-17:
Ok, I'll check the link you shared. Thanks again for your interest @FrancescoTosetti.

amirversal, 2022-11-25:
Hi, thank you for sharing, getting same error, any news?

FrancescoTosetti, 2022-11-25:
Hi @amirversal, I can confirm that the problem occurs when the .h5 graph consists of multiple functional modules. Unfortunately, redefining the network from scratch is not always easy, and you need the model definition code.
We need @Xilinx to fix this
@renjiangyong 

Francesco

FrancescoTosetti, 2022-12-05:
Hi @AbdoSaeed, no I am sorry, but I don't know how to help you. Another option instead of flattening the model is re-defining it from scratch, layer by layer, then training the model again. Depending on your network, it could take a long time. The best option remains to wait for Xilinx for bug fixing. @renjiangyong

FrancescoTosetti, 2022-12-09:
@AbdoSaeed let us know if you are able to compile the model as well. By the way, I managed to quantize and compile flat models, but it's not feasible to do this for each model we want to accelerate with Vitis AI. Now that the cause of the problem is confirmed, let's hope that Xilinx will fix the bug ASAP. 

@renjiangyong can you provide some news on this?

Thank you

Francesco

AbdoSaeed, 2022-12-11:
Yes, I managed to compile the model

gengxinwu, 2023-04-20:
Hi @FrancescoTosetti. As discussed above, the reason for this issue is that vai_q_tensorflow2 does not yet support submodel. For a workaround, we can flatten submodel and reload its weights. please refer to [#1088](https://github.com/Xilinx/Vitis-AI/issues/1088) for a detailed example.

I understand that the upcomming version of vai_q_tensorflow2 will support quantizing nested submodels, but the compiler still needs some time to support this. 

Thank you.

renjiangyong, 2023-06-21:
This issue has been inactive for 2 months, so we will close it for now. If you have any further concerns or need assistance, please feel free to reopen it at any time.

closed 2023-06-21 by renjiangyong