Xilinx/Vitis-AI #952: hal_host_trace.csv missing when profile the model

AdrFebles, 2022-08-04:
Hi!
When I use vaitrace  to profile the model. I got the following files:

`python3 -m vaitrace_py ./app.py 2 /usr/share/vitis_ai_library/models/PATH DEL XMODEL/`

summary.csv, profile_summary.csv, vart_trace.csv, xrt.run_summary and vitis_ai_profile.csv but I don't got the hal_host_trace.csv, so the Timiline Trace in Vitis Analyzer shows only this:
[image]

Does anyone have any idea what might be going wrong?
Thanks!

qianglin-xlnx, 2022-08-31:
Hi @AdrFebles 
Thank you for your feedback. We have filed a Jira internal. If there are any updates, I will let you know. Thank you.

AdrFebles, 2022-09-01:
> Hi @AdrFebles Thank you for your feedback. We have filed a Jira internal. If there are any updates, I will let you know. Thank you.

Ok @qianglin-xlnx, thank you

li900309, 2022-09-26:
hi @AdrFebles  as far as I know, hal_host_trace.csv seems to be obsolete by xrt after 21.1, What information do you need from this

AdrFebles, 2022-09-30:
> hi @AdrFebles as far as I know, hal_host_trace.csv seems to be obsolete by xrt after 21.1, What information do you need from this

Ok @li900309, I didn't have this information. With the rest of .csv files I managed to analyze the DPU Performance, but the CPU usage is not shown.
Thanks for replying

closed 2022-10-14 by qianglin-xlnx