Xilinx/Vitis-AI #955: Hello! problems during installing VART

HankIC, 2022-08-05:
Hello! I'm installing VART v1.1. I want to set up the host for edge. I follow the readme file to the final step: bash -x build.sh. I meet this problem:
build.sh: line 27: aarch64-xilinx-linux-g++: command not found
How can I fix this problem? Thank you!
[image]

HankIC, 2022-08-05:
[image]

lishixlnx, 2022-08-11:
as the readme mentions:
 If you encounter any path errors in running examples, check to see if you follow the steps above to set the host.
 
 Please follow the steps carefully.
did you run the steps 
# ./host_cross_compiler_setup.sh
mentioned in below page ?
https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc#step1-setup-cross-compiler

HankIC, 2022-08-11:
> as the readme mentions: If you encounter any path errors in running examples, check to see if you follow the steps above to set the host.
>
> Please follow the steps carefully. did you run the steps
>
> # [...]

Thank you friend! I reboot the computer, rerun the "unset LD_LIBRARY_PATH; source ~/petalinux_sdk/environment-setup-aarch64-xilinx-linux" command, and rerun "bash -x [build.sh](https://build.sh/)" again according to the “https://github.com/Xilinx/Vitis-AI/tree/1.1/VART”. It works!

closed 2022-08-16 by qianglin-xlnx