Xilinx/Vitis-AI #956: What's the difference between _ana() and _parallel_ana() in pytorch iterative pruning method?

LorenzoSun-V, 2022-08-08:
I use torch DDP to train my model, but can also use single GPU to do eval_fn() when analyzing.  I follow up the ana source code  and found out that the difference is between _ana() and _parallel_ana() but using DDP and normal mode to infer the model. So, what's the difference between _ana() and _parallel_ana() in pytorch iterative pruning method?

yuwang-xlnx, 2022-08-15:
Hi @LorenzoSun-V,

_ana() use single GPU to do model analysis, while _parallel_ana() tries to use all available GPUs to do it.

LorenzoSun-V, 2022-08-18:
In this [reply](https://github.com/Xilinx/Vitis-AI/issues/957#issuecomment-1217681909), the program seems to use only one gpu to analyse. Is that normal?

wangywy, 2022-09-07:
@LorenzoSun-V 
Theoretically that is abnormal. Could it be an environmental issue?