Xilinx/Vitis-AI #957: A question in iterative pruning by pytorch.

LorenzoSun-V, 2022-08-09:
When I reproduce [cifar10_ddp](https://github.com/Xilinx/Vitis-AI/tree/master/examples/Vitis-AI-Optimizer/vai_p_pytorch/cifar10_ddp)  example, the program stuck in the last step of analyse so I must kill the program manually. The stuck program was like:

[image]

This question is also corresponding to #956 .

ruiyingy1213, 2022-08-17:
Hi, can you tell how do you install the tool, with docker or other ways?

LorenzoSun-V, 2022-08-17:
Hi @ruiyingy1213 .
I install the pruning tools by docker and get the trial license from FAE of Xilinx .

ruiyingy1213, 2022-08-17:
Which docker container do you use?

LorenzoSun-V, 2022-08-17:
I use this [script](https://github.com/Xilinx/Vitis-AI/blob/master/docker/docker_build_gpu.sh) to build my docker env.

ruiyingy1213, 2022-08-17:
Have you tried to use the pre-built docker container? https://hub.docker.com/r/xilinx/vitis-ai/tags

LorenzoSun-V, 2022-08-17:
No, in fact I follow [README](https://github.com/Xilinx/Vitis-AI/blob/master/README.md) to setup my Vitis-AI docker env. Maybe you can try this again. Moreover, you should export the pruning license in the docker env before using pruning tools.

ruiyingy1213, 2022-08-17:
Ok, I will try.

ruiyingy1213, 2022-08-17:
Hi, I run the following steps：

cd docker
./docker_build_gpu.sh
./docker_run.sh xilinx/vitis-ai-gpu:latest
cd examples/Vitis-AI-Optimizer/vai_p_pytorch/cifar10_ddp
bash 1_train_baseline_ddp.sh
bash 2_analyse.sh

And everything is okay.

LorenzoSun-V, 2022-08-17:
OK，I'll try again. I used the pretrained weight and skipped baseline training.

LorenzoSun-V, 2022-08-17:
Hi @ruiyingy1213 , I've tried to run analyse script again and it run correctly. In this [reply](https://github.com/Xilinx/Vitis-AI/issues/956#issuecomment-1214783153), _parallel_ana() will use muti-gpus to analyse. However, the program seems to occupy only one gpu.
[image]

I use another computer to run this script serveral times and sometimes the program encountered stuck problems. So I use CTRL+C to terminal the program manually and after that the program will generate the xxx.sens fiel in .vai folder.

ruiyingy1213, 2022-08-17:
Can you locate the line of code where it stuck?

LorenzoSun-V, 2022-08-18:
It's hard to locate the stuck code actully because the main code is stuck here.
[image]

I'm more inclined to blame the problem on multiprocessing. Concretly, ana() code is:
[image]

And AnaTask is:
[image]

The print info incudes "I0817 08:21:45.822546 structured.py] Analysis complete 37/37", and then stuck here. After CTRL+C the code generated the xxx.sens file. Maybe one of process was stuck or asynchronize with others so that the program was stuck.

Qiang-Feng98, 2023-08-07:
Hi, When I prune my model, I meet an error: **_nndct_shared.pruning.errors.OptimizerNotExcludeNodeError: Must exclude node from pruning: FastCS::FastCS/Sequential[decoder]/ConvTranspose2d[0]/ret.7._**
I tried to solve this with the excludes parameter: **_excludes = ['FastCS::FastCS/Sequential[decoder]/ConvTranspose2d[0]/ret.7' ], pruing_runner.ana(eval_fn, gpus, excludes=excludes)_**
but it didn't work. How should I use the excludes parameter?
I used Vitis-AI 3.0 on Ubuntu 20.04. Thanks.

ruiyingy1213, 2023-08-07:
excludes: A list of node name or torch module to be excluded from pruning.
It seems the usage is right.

Qiang-Feng98, 2023-08-07:
@ruiyingy1213 
Thanks for your reply.
This can't work. I don't know why.
Is there any other solution?
Thanks again.

ruiyingy1213, 2023-08-07:
What error is reported？

Qiang-Feng98, 2023-08-07:
**_nndct_shared.pruning.errors.OptimizerNotExcludeNodeError: Must exclude node from pruning: FastCS::FastCS/Sequential[decoder]/ConvTranspose2d[0]/ret.7._**
This error is not resolved. Thanks again.

ruiyingy1213, 2023-08-07:
Maybe you can try to exclude the node by torch module object. If it still doesn't work, you can provide the code script and related info required by code run. I will replicate the problem and try to solve.

Qiang-Feng98, 2023-08-07:
Sorry, I don't have a solution for this, although it seems relatively simple. I've uploaded the script, trained model and other files,  it runs directly in the vitis-ai-pytorch environment. Thanks again for your help.

链接：[](https://pan.baidu.com/s/1tUnLIlnhbnW79M72NzkC-g )
提取码：7r59

can you get it?

ruiyingy1213, 2023-08-07:
I don‘t  get the code link.

Qiang-Feng98, 2023-08-07:
Sorry, I don't have a solution for this, although it seems relatively simple. I've uploaded the script, trained model and other files, it runs directly in the vitis-ai-pytorch environment.
[Xilinx.zip](https://github.com/Xilinx/Vitis-AI/files/12272916/Xilinx.zip)

ruiyingy1213, 2023-08-07:
I have replicated the problem and try to solve.

ruiyingy1213, 2023-08-07:
I solve it by excluding the below node:
excludes = ['FastCS::FastCS/Sequential[rec2]/PartialConv[0]/Conv2d[partial_conv3]/ret.17']
For concat operation, there are some usage limitation. Maybe the error message is wrong.

Qiang-Feng98, 2023-08-07:
Oh, thank you so much for your help! 
Can I ask how you got this correct error message?
Thank you very much.

ruiyingy1213, 2023-08-07:
For concat operation, if one of the inputs is not been pruned and one of the inputs is been pruned, the error message that the prunable ancestor of the pruned input must be excluded will be triggered according to the logic of  the current code.  As my understanding，the  prunable ancestor has nothing to do with the current concat op. Maybe you can simply think that pruning of concate op's input op is not supported.

Qiang-Feng98, 2023-08-07:
Oh, Thank you very much.