Xilinx/Vitis-AI #958: Tensorflow2 Compilation Error

kkenshin1, 2022-08-09:
When I use Vitis-AI 1.4  to compile my own model, there are some errors below.
```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['./my_quantized.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/my_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/experiment/my_model/train_tf2/my_quantized.h5
[INFO] keras version: 2.4.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|█████████████████████| 17/17 [00:00<00:00, 24957.36it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████| 30/30 [00:00<00:00, 28339.89it/s]
[INFO] perform level-0 opt :100%|█████████████████████████| 2/2 [00:00<00:00, 487.79it/s]
[INFO] perform level-1 opt :100%|█████████████████████████| 2/2 [00:00<00:00, 1684.80it/s]
[INFO] infer shape (NHWC)  :100%|████████████████████████| 34/34 [00:00<00:00, 42429.73it/s]
[INFO] generate xmodel     :100%|████████████████| 34/34 [00:00<00:00, 11385.74it/s]
[INFO] dump xmodel: /tmp/my_org.xmodel
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Graph name: functional_1, with op num: 40
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_UNSUPPORT_QUANTIZATION][The fix info is error or unsupported.] elew1's quantization info's variation is too large.
*** Check failure stack trace: ***
Aborted (core dumped)
```

That shows that there are some errors when I quantize my model. And below is my quantization code.
```
    model = load_model("./my.h5")
    quantizer = vitis_quantize.VitisQuantizer(model)
    quantized_model = quantizer.quantize_model(calib_dataset=[x1,x2])
    quantized_model.save('my_quantized.h5')
```

But When I want to dump the weight and bias quantized data, the output folder is empty.
```
quantizer.dump_model(quantized_model, dataset=[x1_dump,x2_dump],weights_only=True)
```

Is there any solutions for this error?

Thanks!!

Vitis-AI-4836, 2022-08-11:
hi kkenshin1,
  Please check your elew's shift_write value which bases on maximum difference between input and output fix-point value.  Our shift_write value range is limited to [-7, 7].

kkenshin1, 2022-08-11:
Cool, thanks! I have solved my probem.

Another question, if I want to integrate the DPU IP into my own vivado/vitis project, and is there any documentation that can help?

Vitis-AI-4836, 2022-08-16:
We only have .xo release for DPU, refer to PG403, you can find it on XILINX official website.

kkenshin1, 2022-08-16:
Okay, thanks for your help.

closed 2022-08-16 by kkenshin1