Xilinx/Vitis-AI #959: PyTorch Mnist example

Werterror, 2022-08-09:
Hi all,
I tired running Vitis-AI-Tutorial:
https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/09-mnist_pyt/README.md
And while running code from the Readme file, software tries to download files from page that no longer exists:
http://yann.lecun.com/exdb/mnist/https://ossci-datasets.s3.amazonaws.com/mnist/train-images-idx3-ubyte.gz
More details on screenshot:
[image]

qianglin-xlnx, 2022-08-10:
Hi @Werterror 
Maybe you can try this https://ossci-datasets.s3.amazonaws.com/mnist/train-images-idx3-ubyte.gz.

Prithvi-Velicheti, 2022-08-15:
@qianglin-xlnx can you elaborate the solution ? you mean should i manually download them and keep in the location.

lishixlnx, 2022-08-16:
please change the train.py to below and try again:

 **40 torchvision.datasets.MNIST.resources = [
 41     # ('https://ossci-datasets.s3.amazonaws.com/mnist/train-images-idx3-ubyte.gz', 'f68b3c2dcbeaaa9fbdd348bbdeb94873'),
 42     # ('https://ossci-datasets.s3.amazonaws.com/mnist/train-labels-idx1-ubyte.gz', 'd53e105ee54ea40749a09fcbcd1e9432'),
 43     # ('https://ossci-datasets.s3.amazonaws.com/mnist/t10k-images-idx3-ubyte.gz', '9fb629c4189551a2d022fa330f9573f3'),
 44     # ('https://ossci-datasets.s3.amazonaws.com/mnist/t10k-labels-idx1-ubyte.gz', 'ec29112dd5afa0611ce80d1b7f02629c')
 45     ('train-images-idx3-ubyte.gz', 'f68b3c2dcbeaaa9fbdd348bbdeb94873'),
 46     ('train-labels-idx1-ubyte.gz', 'd53e105ee54ea40749a09fcbcd1e9432'),
 47     ('t10k-images-idx3-ubyte.gz', '9fb629c4189551a2d022fa330f9573f3'),
 48     ('t10k-labels-idx1-ubyte.gz', 'ec29112dd5afa0611ce80d1b7f02629c')
 49 ]**

Werterror, 2022-08-17:
> please change the train.py to below and try again:
>
> **40 torchvision.datasets.MNIST.resources = [ 41 # ('https://ossci-datasets.s3.amazonaws.com/mnist/train-images-idx3-ubyte.gz', [...]

That works, Thank you

closed 2022-08-17 by Werterror