Xilinx/Vitis-AI #960: Error while building gpu docker - manifest not found, no image vitis-ai-gpu

Werterror, 2022-08-09:
Hi all,
I get error while building gpu docker, CUDA version:
`(base) hyperworks2@hyperworks2-MS-7B17:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release 10.1, V10.1.243
(base) hyperworks2@hyperworks2-MS-7B17:~$ /usr/local/cuda/bin/nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Fri_Dec_17_18:16:03_PST_2021
Cuda compilation tools, release 11.6, V11.6.55
Build cuda_11.6.r11.6/compiler.30794723_0
(base) hyperworks2@hyperworks2-MS-7B17:~$ 
`

Return of sudo ./docker_build_gpu.sh:
`(base) hyperworks2@hyperworks2-MS-7B17:~/Vitis-AI/docker$ sudo ./docker_build_gpu.sh
NOTICE:  BY INVOKING THIS SCRIPT AND USING THE SOFTWARE INSTALLED BY THE
SCRIPT, YOU AGREE ON BEHALF OF YOURSELF AND YOUR EMPLOYER (IF APPLICABLE)
TO BE BOUND TO THE LICENSE AGREEMENTS APPLICABLE TO THE SOFTWARE THAT YOU
INSTALL BY RUNNING THE SCRIPT.

BY ELECTING TO CONTINUE, YOU WILL CAUSE THIS SCRIPT FILE TO AUTOMATICALLY
INSTALL A VARIETY OF SOFTWARE COPYRIGHTED
BY XILINX AND THIRD PARTIES THAT IS SUBJECT TO VARIOUS LICENSE AGREEMENTS 
THAT APPEAR UPON INSTALLATION, ACCEPTANCE AND/OR ACTIVATION OF THE
SOFTWARE AND/OR ARE CONTAINED OR DESCRIBED IN THE CORRESPONDING RELEASE
NOTES OR OTHER DOCUMENTATION OR HEADER OR SOURCE FILES. XILINX DOES NOT
GRANT TO LICENSEE ANY RIGHTS OR LICENSES TO SUCH THIRD-PARTY SOFTWARE.
LICENSEE AGREES TO CAREFULLY REVIEW AND ABIDE BY THE TERMS AND CONDITIONS
OF SUCH LICENSE AGREEMENTS TO THE EXTENT THAT THEY GOVERN SUCH SOFTWARE.

BY ELECTING TO CONTINUE, YOU WILL CAUSE THE FOLLOWING SOFTWARE TO BE DOWNLOADED
AND INSTALLED ON YOUR SYSTEM. BY ELECTING TO CONTINUE, YOU UNDERSTAND THAT THE
INSTALLATION OF THE SOFTWARE LISTED BELOW MAY ALSO RESULT IN THE INSTALLATION
ON YOUR SYSTEM OF ADDITIONAL SOFTWARE NOT LISTED BELOW IN ORDER TO OPERATE
(SUCH SOFTWARE IS HEREAFTER REFERRED TO AS ‘DEPENDENCIES’)
XILINX DOES NOT GRANT TO LICENSEE ANY RIGHTS OR LICENSES TO SUCH DEPENDENCIES
LICENSEE AGREES TO CAREFULLY REVIEW AND ABIDE BY THE TERMS AND CONDITIONS
OF ANY LICENSE AGREEMENTS TO THE EXTENT THAT THEY GOVERN SUCH DEPENDENCIES

BY ELECTING TO CONTINUE, YOU WILL CAUSE THE FOLLOWING SOFTWARE PACKAGES
(AND THEIR RESPECTIVE DEPENDENCIES, IF APPLICABLE) TO BE DOWNLOADED FROM
UBUNTU'S MAIN REPO AND INSTALLED ON YOUR SYSTEM:
http://us.archive.ubuntu.com/ubuntu/dists/bionic/
http://us.archive.ubuntu.com/ubuntu/dists/bionic/
1.  sudo  
2.  git  
3.  zstd  
4.  tree  
5.  vim  
6.  wget  
7.  bzip2  
8.  ca-certificates  
9.  curl  
10. unzip  
11. python3-minimal  
12. python3-opencv  
13. python3-venv  
14. python3-pip  
15. python3-setuptools  
16. g++  
17. make  
18. cmake  
19. build-essential  
20. autoconf  
21. libgoogle-glog-dev  
22. libgflags-dev  
23. libunwind-dev  
24. libtool  
25. libgtk2.0-dev
26. libavcodec-dev
27. libavformat-dev
28. libavdevice-dev

BY ELECTING TO CONTINUE, YOU WILL CAUSE THE FOLLOWING SOFTWARE PACKAGES
(AND THEIR RESPECTIVE DEPENDENCIES, IF APPLICABLE) TO BE DOWNLOADED FROM
ANACONDA REPO AND INSTALLED ON YOUR SYSTEM:
https://anaconda.org”
1.  absl-py
2.  astor
3.  attrs
4.  backcall
5.  backports
6.  backports.weakref
7.  blas
8.  bleach
9.  boost
10.  bzip2
11.  ca-certificates
12.  cairo
13.  c-ares
14.  certifi
15.  cffi
16.  chardet
17.  cloudpickle
18.  conda
19.  conda-package-handling
20.  cryptography
21.  cycler
22.  cytoolz
23.  dask-core
24.  dbus
25.  decorator
26.  defusedxml
27.  dill
28.  dpuv1_compiler
29.  dpuv1-rt
30.  dpuv1-rt-ext
31.  dpuv1-rt-neptune
32.  entrypoints
33.  expat
34.  ffmpeg
35.  fontconfig
36.  freeglut
37.  freetype
38.  fribidi
39.  gast
40.  gettext
41.  gflags
42.  giflib
43.  glib
44.  glog
45.  gmp
46.  gnutls
47.  google-pasta
48.  graphite2
49.  graphviz
50.  grpcio
51.  gst-plugins-base
52.  gstreamer
53.  h5py
54.  harfbuzz
55.  hdf5
56.  icu
57.  idna
58.  imageio
59.  importlib_metadata
60.  importlib-metadata
61.  intel-openmp
62.  ipykernel
63.  ipython
64.  ipython_genutils
65.  ipywidgets
66.  jasper
67.  jedi
68.  jinja2
69.  joblib
70.  jpeg
71.  json-c
72.  jsoncpp
73.  jsonschema
74.  jupyter
75.  jupyter_client
76.  jupyter_console
77.  jupyter_core
78.  keras
79.  keras-applications
80.  keras-base
81.  keras-preprocessing
82.  kiwisolver
83.  krb5
84.  lame
85.  ld_impl_linux-64
86.  leveldb
87.  libblas
88.  libboost
89.  libcblas
90.  libedit
91.  libffi
92.  _libgcc_mutex
93.  libgcc-ng
94.  libgfortran-ng
95.  libglu
96.  libiconv
97.  liblapack
98.  liblapacke
99.  libopenblas
100.  libopencv
101.  libopus
102.  libpng
103.  libprotobuf
104.  libsodium
105.  libssh2
106.  libstdcxx-ng
107.  libtiff
108.  libtool
109.  libuuid
110.  libvpx
111.  libwebp
112.  libxcb
113.  libxml2
114.  lmdb
115.  lz4-c
116.  markdown
117.  markupsafe
118.  marshmallow
119.  matplotlib
120.  matplotlib-base
121.  mistune
122.  mkl
123.  mkl_fft
124.  mkl_random
125.  mkl-service
126.  mock
127.  more-itertools
128.  nbconvert
129.  nbformat
130.  ncurses
131.  nettle
132.  networkx
133.  notebook
134.  numpy
135.  numpy-base
136.  olefile
137.  openblas
138.  opencv
139.  openh264
140.  openssl
141.  opt_einsum
142.  packaging
143.  pandas
144.  pandoc
145.  pandocfilters
146.  pango
147.  parso
148.  pexpect
149.  pickleshare
150.  pillow
151.  pip
152.  pixman
153.  pluggy
154.  progressbar2
155.  prometheus_client
156.  prompt_toolkit
157.  prompt-toolkit
158.  protobuf
159.  ptyprocess
160.  py
161.  pybind11
162.  py-boost
163.  pycosat
163.  pycosat
164.  pycparser
165.  pydot
166.  pygments
167.  py-opencv
168.  pyopenssl
169.  pyparsing
170.  pyqt
171.  pyrsistent
172.  pysocks
173.  pytest
174.  pytest-runner
175.  python
176.  python-dateutil
177.  python-gflags
178.  python-graphviz
179.  python-leveldb
180.  python-utils
181.  pytz
182.  pywavelets
183.  pyyaml
184.  pyzmq
185.  qt
186.  qtconsole
187.  qtpy
188.  readline
189.  requests
190.  ruamel_yaml
191.  scikit-image
192.  scikit-learn
193.  scipy
194.  send2trash
195.  setuptools
196.  sip
197.  six
198.  snappy
199.  sqlite
200.  tensorboard
201.  tensorflow
202.  tensorflow-base
203.  tensorflow-estimator
204.  termcolor
205.  terminado
206.  testpath
207.  _tflow_select
208.  threadpoolctl
209.  tk
210.  toolz
211.  tornado
212.  tqdm
213.  traitlets
214.  urllib3
215.  wcwidth
216.  webencodings
217.  werkzeug
218.  wheel
219.  widgetsnbextension
220.  wrapt
221.  x264
222.  xcompiler
223.  xorg-libice
224.  xorg-libsm
225.  xorg-libx11
226.  xorg-libxext
227.  xorg-libxpm
228.  xorg-libxrender
229.  xorg-libxt
230.  xorg-renderproto
231.  xorg-xextproto
232.  xorg-xproto
233.  xz
234.  yaml
235.  yaml-cpp
236.  zeromq
237.  zipp
238.  zlib
239.  zstd

BY ELECTING TO CONTINUE, YOU ACKNOWLEDGE AND AGREE, FOR YOURSELF AND ON BEHALF
OF YOUR EMPLOYER (IF APPLICABLE), THAT XILINX IS NOT DISTRIBUTING TO YOU IN
THIS FILE ANY OF THE AFORMENTIONED SOFTWARE OR DEPENDENCIES, AND THAT YOU ARE
SOLELY RESPONSIBLE FOR THE INSTALLATION OF SUCH SOFTWARE AND DEPENDENCIES ON
YOUR SYSTEM AND FOR CAREFULLY REVIEWING AND ABIDING BY THE TERMS AND CONDITIONS
OF ANY LICENSE AGREEMENTS TO THE EXTENT THAT THEY GOVERN SUCH SOFTWARE AND DEPENDENCIES

Press any key to continue...

Do you agree to the terms and wish to proceed [y/n]? 

Do you agree to the terms and wish to proceed [y/n]? y
Sending build context to Docker daemon  86.02kB
Step 1/61 : FROM nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04
manifest for nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04 not found: manifest unknown: manifest unknown
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.5.0.1260
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.5.0.1260
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.5.0.1260
`

janifer112x, 2022-08-11:
Hi @Werterror ,
this error is due to Nvidia updated their docker tag in docker hub, the base image we are using does not exist any more, we just updated the version from 11.0 to 11.0.3, Could you please pull the latest commit from master branch and try again?

Prithvi-Velicheti, 2022-08-14:
janifer can you please elaborate the answer , I got the same answer.

Prithvi-Velicheti, 2022-08-14:
thanks, seems to be resolved by changing vitis-ai-gpu.Dockerfile

Werterror, 2022-08-16:
> Hi @Werterror , this error is due to Nvidia updated their docker tag in docker hub, the base image we are using does not exist any more, we just updated the version from 11.0 to 11.0.3, Could you please pull the latest commit from master branch and try again?

It does build now without any errors but when I try to run it it returns error:
`docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].`
I didnt find any other Vits-AI related solution then to instal nvidia container-toolkit, but that solution wasnt confirmed.
Is there a known solution?
I have CUDA ready GPU and CUDA 11.6 installed.

Werterror, 2022-08-16:
I found another thread confirming that installation of nvidia container-toolkit solves the problem, so I went through with it and gpu docker works just fine. Here's installation guide: https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html
Thank you for your help.

closed 2022-08-16 by Werterror