Xilinx/Vitis-AI #961: Questions about deploying coco yolox

jzymessi, 2022-08-10:
I trained the coco dataset according to the Yolox model in the Model zoo. During the deployment, I found that there was a problem with the output bbox. When I wanted to check the specific code of post-processing, I only traced to this side, no specific implementation function was found
[image]
How do I solve it?

zz002, 2022-08-10:
@jzymessi https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Library/xnnpp/src/yolovx.cpp#L69

jzymessi, 2022-08-11:
@zz002 ：
     thanks, I modified some code in master/src/Vitis-AI-Library/ .then run ./cmake.sh --clean compile .so.
     But it does't work, how to slove it?

closed 2022-08-31 by qianglin-xlnx