Xilinx/Vitis-AI #962: facenet-pytorch pruning error

jia0511, 2022-08-10:
Hello xilinx,

I have a problem with weight matching when using vitis-ai pruning：

**重现步骤**

1.下载配置vitis-vi2.5 docker环境  

https://github.com/Xilinx/Vitis-AI/blob/master/docs/quick-start/install/install_docker/README.md

2、下载vitis-ai环境包
git clone --recurse-submodules https://github.com/Xilinx/Vitis-AI  

3、下载facenet-pytorch版本代码  
链接：https://pan.baidu.com/s/1X6SDMWzri03ZS7LcuUV5ig 
提取码：r7i8 

4.下载训练webface和测试数据集lfw  
链接：https://pan.baidu.com/s/19SkulNP7LSLkRc-KUW1rNA
提取码：pfbs

5、进入docker  
sh run.sh

6、激活conda剪枝环境进入facenet目录  
conda activate pytorch_for_facenet  
cd facenet-pytorh

7、 运行一次剪枝search脚本  
bash 2_search.sh

8、运行迭代剪枝脚本 （已经完成分析）  
bash 3_retrain_sparse_model.sh

<br>

**测试结果**

1.一次剪枝遇到的问题  
(vitis-ai-optimizer\_pytorch) Vitis-AI /workspace/facenet-pytorch > bash 2_search.sh

\[VAIQ\_NOTE\]: Loading NNDCT kernels...  
Load weights model\_data/facenet\_inception\_resnetv1.pth.

\[VAIQ\_NOTE\]: Start to trace model...

\[VAIQ\_NOTE\]: Finish tracing.

\[VAIQ\_NOTE\]: Processing ops...  
██████████████████████████████████████████████████| 475/475 \[00:00<00:00, 2591.0

\[VAIQ\_WARN\]: The quantizer recognize new op `expand_as` as a float operator by default.

\[VAIQ\_WARN\]: The quantizer recognize new op `normalize` as a float operator by default.

\[VAIQ\_WARN\]: The quantizer recognize new op `clamp` as a float operator by default.  
98.73  
Traceback (most recent call last):  
File "search.py", line 68, in <module>  
eval\_args=(data\_loader,))  
File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 641, in search  
excluded\_nodes, calib\_args, eval\_args, orig\_macs)  
File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 701, in *search\_random\_precheck&nbsp;pruned\_model,&nbsp;pruned\_graph,&nbsp;pruning\_info&nbsp;=&nbsp;self.*prune(spec, mode='slim') File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 161, in \_prune model = func(pruned\_graph, pruning\_info) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 196, in \_generate\_slim\_model pruned\_module = mod\_util.create\_module\_by\_node(type(module), node) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/utils/module\_util.py", line 159, in create\_module\_by\_node module.load\_state\_dict(state\_dict\_from\_node(node)) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1483, in load\_state\_dict self.\_\_class**.**name\*\*, "\\n\\t".join(error\_msgs)))  
RuntimeError: Error(s) in loading state\_dict for Conv2d:  
size mismatch for weight: copying a param with shape torch.Size(\[710, 162, 1, 1\]) from checkpoint, the shape in current model is torch.Size(\[198, 162, 1, 1\]).  
size mismatch for bias: copying a param with shape torch.Size(\[710\]) from checkpoint, the shape in current model is torch.Size(\[198\]).  
(vitis-ai-optimizer\_pytorch) Vitis-AI /workspace/facenet-pytorch >

2.迭代剪枝遇到的问题  
(vitis-ai-optimizer\_pytorch) Vitis-AI /workspace/facenet-pytorch > bash 3_retrain_sparse_model.sh

\[VAIQ\_NOTE\]: Loading NNDCT kernels...  
Load weights model\_data/facenet\_inception\_resnetv1.pth.

\[VAIQ\_NOTE\]: Start to trace model...

\[VAIQ\_NOTE\]: Finish tracing.

\[VAIQ\_NOTE\]: Processing ops...  
██████████████████████████████████████████████████| 475/475 \[00:00<00:00, 2561.2

\[VAIQ\_WARN\]: The quantizer recognize new op `normalize` as a float operator by default.

\[VAIQ\_WARN\]: The quantizer recognize new op `expand_as` as a float operator by default.

\[VAIQ\_WARN\]: The quantizer recognize new op `clamp` as a float operator by default.  
Input type (torch.FloatTensor) and weight type (torch.cuda.FloatTensor) should be the same or input should be a MKLDNN tensor and weight is a dense tensor  
Error occurs, No graph saved  
Traceback (most recent call last):  
File "sparse\_model\_train.py", line 98, in <module>  
model = pruning\_runner.prune(removal\_ratio=args.sparsity, mode='sparse',excludes = \[model.backbone.model.block8.conv2d\])  
File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 426, in prune  
spec = self.*spec\_from\_ratio(net\_sens,&nbsp;removal\_ratio,&nbsp;excluded\_nodes)&nbsp;File&nbsp;"/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py",&nbsp;line&nbsp;500,&nbsp;in&nbsp;*spec\_from\_ratio pruned\_model = self.\_prune(cur\_spec, 'slim')\[0\] File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 161, in \_prune model = func(pruned\_graph, pruning\_info) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/pruning/structured.py", line 196, in \_generate\_slim\_model pruned\_module = mod\_util.create\_module\_by\_node(type(module), node) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/pytorch\_nndct/utils/module\_util.py", line 159, in create\_module\_by\_node module.load\_state\_dict(state\_dict\_from\_node(node)) File "/opt/vitis\_ai/conda/envs/vitis-ai-optimizer\_pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1483, in load\_state\_dict self.\_\_class**.**name\*\*, "\\n\\t".join(error\_msgs)))  
RuntimeError: Error(s) in loading state\_dict for Conv2d:  
size mismatch for weight: copying a param with shape torch.Size(\[552, 28, 1, 1\]) from checkpoint, the shape in current model is torch.Size(\[40, 28, 1, 1\]).  
size mismatch for bias: copying a param with shape torch.Size(\[552\]) from checkpoint, the shape in current model is torch.Size(\[40\]).  
(vitis-ai-optimizer\_pytorch) Vitis-AI /workspace/facenet-pytorch >

<br>

**期望结果**

1.一次剪枝权重正常匹配

2.迭代剪枝权重正常匹配

DehuaTang, 2022-08-10:
Can you show me the code of search.py ? There is no problem with this step of model analysis ?

jia0511, 2022-08-10:
hello @DehuaTang 

Analysis did not encounter problems，search.py code

import os
import argparse
import numpy as np
import torch
from torch.utils.data import DataLoader
from utils.utils import get_num_classes, show_config
from nets.facenet import Facenet
from utils.dataloader import FacenetDataset, LFWDataset, dataset_collate

from pytorch_nndct import get_pruning_runner
from prune_utils import *            

parser = argparse.ArgumentParser()
parser.add_argument(
    '--gpus', type=str, default='0', help='String of available GPU number')
parser.add_argument(
    '--model_path',
    type=str,
    default='model_data/facenet_inception_resnetv1.pth',
    help='Pretrained model filepath')
parser.add_argument(
    '--backbone',
    type=str,
    default='inception_resnetv1',
    help='network backbone')
parser.add_argument(
    '--lfw_pairs_path',
    type=str,
    default='model_data/lfw_pair.txt',
    help='lfw pairs path')
parser.add_argument(
    '--annotation_path',
    type=str,
    default='cls_train.txt',
    help='annotation_path')    
parser.add_argument(
    '--num_subnet', type=int, default=20, help='The number of subnets searched')
parser.add_argument(
    '--sparsity', type=float, default=0.5, help='Sparsity ratio')
    
parser.add_argument('--batch_size', type=int, default=32, help='Batch size')
args, _ = parser.parse_known_args()

gpus = get_gpus(args.gpus)

                    
if __name__ == "__main__":
    lfw_dir_path    = "lfw"
    
    num_classes = get_num_classes(args.annotation_path)
    model = Facenet(backbone=args.backbone, num_classes=num_classes, pretrained=True)
    load_weight(model, args.model_path)
        
    input_shape     = [160, 160, 3]    
    data_loader = torch.utils.data.DataLoader(
        LFWDataset(dir=lfw_dir_path, pairs_path=args.lfw_pairs_path, image_size=input_shape), batch_size=32, shuffle=False)
    
    input_signature = torch.randn([1, 3, 160, 160], dtype=torch.float32)
    pruning_runner = get_pruning_runner(model, input_signature, 'one_step')                                     
    
    pruning_runner.search(
        gpus=gpus,
        calibration_fn=calibration_fn,
        calib_args=(data_loader,),
        num_subnet=args.num_subnet,
        removal_ratio=args.sparsity,
        eval_fn=eval_fn,
        eval_args=(data_loader,))
      
    print('Search done!')  

prune_utils.py code is follow:
import os

import numpy as np
import torch
from utils.utils_metrics import evaluate

def get_gpus(device):
  return [int(i) for i in device.split(',')]
  
def calibration_fn(model, test_loader, number_forward=100):
  model.train()
  print("Adaptive BN atart...")
  with torch.no_grad():
    for index, (data_a, data_p, label) in enumerate(test_loader):
      data_a, data_p = data_a.type(torch.FloatTensor), data_p.type(torch.FloatTensor)
      data_a, data_p = data_a.cuda(), data_p.cuda()
      model(data_a), model(data_p)
      if index > number_forward:
        break
  print("Adaptive BN end...")
  
def eval_fn(model, test_loader):
    model.eval()
    labels, distances = [], []
    for _, (data_a, data_p, label) in enumerate(test_loader):
        with torch.no_grad():
            data_a, data_p = data_a.type(torch.FloatTensor), data_p.type(torch.FloatTensor)
            data_a, data_p = data_a.cuda(), data_p.cuda()
            out_a, out_p = model(data_a), model(data_p)
            dists = torch.sqrt(torch.sum((out_a - out_p) ** 2, 1))
        distances.append(dists.data.cpu().numpy())
        labels.append(label.data.cpu().numpy())
        
    labels      = np.array([sublabel for label in labels for sublabel in label])
    distances   = np.array([subdist for dist in distances for subdist in dist])
    _, _, accuracy, _, _, _, _ = evaluate(distances,labels)
    #print('LFW_Accuracy: %2.5f+-%2.5f' % (np.mean(accuracy), np.std(accuracy)))
    #acc = round(np.mean(accuracy*100), 5)
    print(accuracy)
    return accuracy

def load_weight(model, model_path):
    print('Load weights {}.'.format(model_path))
    
    #------------------------------------------------------#
    #   根据预训练权重的Key和模型的Key进行加载
    #------------------------------------------------------#
    model_dict      = model.state_dict()
    pretrained_dict = torch.load(model_path, map_location = 'cuda')
    load_key, no_load_key, temp_dict = [], [], {}
    for k, v in pretrained_dict.items():
        if k in model_dict.keys() and np.shape(model_dict[k]) == np.shape(v):
            temp_dict[k] = v
            load_key.append(k)
        else:
            no_load_key.append(k)
    model_dict.update(temp_dict)
    model.load_state_dict(model_dict)

jia0511, 2022-08-10:
Hello @DehuaTang
you can download complete code, from baidu pan:

链接：https://pan.baidu.com/s/1X6SDMWzri03ZS7LcuUV5ig
提取码：r7i8

DehuaTang, 2022-08-13:
The file you uploaded is not correct. There are only 'dataset' 'finetune' 'prune'.

jia0511, 2022-08-15:
Hi @DehuaTang 
Sorry, I also found this problem. At present, the code and data set have been packaged into a complete file and sent to Baidu network disk. You can use it directly after downloading.

链接：https://pan.baidu.com/s/1Cgd8yBx36HEKfQBokKTQ0w 
提取码：v697

DehuaTang, 2022-08-15:
This file is so big that it takes a long time to download. Can you give me only the files related to the code?

jia0511, 2022-08-15:
@DehuaTang 

I have removed the database，the original code reference is here：https://github.com/bubbliiiing/facenet-pytorch.

链接：https://pan.baidu.com/s/1xPce_ZGE1ne82jehl5jWJA 
提取码：6q2i

jia0511, 2022-08-16:
@DehuaTang 

I want to know if you have reproduced the problem, what can I do to solve this problem now？

DehuaTang, 2022-08-19:
We have reproduced the problem and are working on a solution.

jia0511, 2022-08-24:
Hi @DehuaTang

I would like to know if there is a good solution to the current problem？

ruiyingy1213, 2022-08-24:
Hi @jia0511 
You can try this:
excludes = ['Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_2]/Block17[0]/Conv2d[conv2d]/25701', 'Facenet::Facenet/inception_resnet[backbone]/Mixed_6a[model]/Mixed_6a[mixed_6a]/Sequential[bran    ch1]/BasicConv2d[2]/Conv2d[conv]/input.118', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_1]/Block35[4]/Conv2d[conv2d]/25449', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequent    ial[repeat_3]/Block8[0]/Conv2d[conv2d]/27210', 'Facenet::Facenet/inception_resnet[backbone]/Mixed_7a[model]/Mixed_7a[mixed_7a]/Sequential[branch1]/BasicConv2d[1]/Conv2d[conv]/input.259', 'Facenet::Facenet/inception_resnet[backbone]/M    ixed_7a[model]/Mixed_7a[mixed_7a]/Sequential[branch2]/BasicConv2d[2]/Conv2d[conv]/input.267', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_2]/Block17[9]/Conv2d[conv2d]/26880']
...
pruning_runner.search(excludes=excludes, ...)
...
The analysis result is wrong. You may need to re-analyse:
...
pruning_runner.ana(excludes=excludes,...)
...

jia0511, 2022-08-24:
Hi @DehuaTang,

Thank you for your support, but I have some problems using the exclusion code you gave me, the screenshot is shown below：

[image]
[image]

[image]

[image]

jia0511, 2022-08-25:
@DehuaTang @ruiyingy1213 

Have you tried to prune this model, can you help me if I solve the problems I encountered after referring to your solution?

ruiyingy1213, 2022-08-26:
Yes, i have tested  'pruning_runner.search()' method and generate '.vai/*.search' file successfully. You may delete the '.vai' directory and try the process from scratch. And i attach the train.py (i.e.train.txt) script.
[train.txt](https://github.com/Xilinx/Vitis-AI/files/9429642/train.txt)

jia0511, 2022-08-26:
@ruiyingy1213 

Thank you for your support very much，but I still have some problems.I copied your train.txt completely and ran it and deleted the .vai directory, corresponding lfw to the official download path, and the network is not modified for any modification, but there are still the following problems：

[image]

ruiyingy1213, 2022-08-29:
It's really strange. I have accomplished the 'search' procedure and generate the *.search file successfully. I'll try run the case in docker env as yours.
[image]
[Facenet_ratio_0.2.search.txt](https://github.com/Xilinx/Vitis-AI/files/9441182/Facenet_ratio_0.2.search.txt)

ruiyingy1213, 2022-08-29:
@jia0511 Maybe i find the question. Try the following excludes:
excludes = ['Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_2]/Block17[0]/Conv2d[conv2d]/25701', 'Facenet::Facenet/inception_resnet[backbone]/Mixed_6a[model]/Mixed_6a[mixed_6a]/Sequential[branch1]/BasicConv2d[2]/Conv2d[conv]/input.235', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_1]/Block35[4]/Conv2d[conv2d]/25449', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_3]/Block8[0]/Conv2d[conv2d]/27210', 'Facenet::Facenet/inception_resnet[backbone]/Mixed_7a[model]/Mixed_7a[mixed_7a]/Sequential[branch1]/BasicConv2d[1]/Conv2d[conv]/input.517', 'Facenet::Facenet/inception_resnet[backbone]/Mixed_7a[model]/Mixed_7a[mixed_7a]/Sequential[branch2]/BasicConv2d[2]/Conv2d[conv]/input.533', 'Facenet::Facenet/inception_resnet[backbone]/Sequential[model]/Sequential[repeat_2]/Block17[9]/Conv2d[conv2d]/26880']

jia0511, 2022-08-30:
@ruiyingy1213 

I feel very much for your support. At present, this problem has been solved, and one-time pruning and iterative pruning can be performed normally.

jia0511, 2022-08-30:
@ruiyingy1213 

I may also need to trouble you at the end. There is no problem in using LFW data to calculate the score during the search and analysis process, but when I evaluate the model with LFW after pruning, the following problem is reported：
[image]

ruiyingy1213, 2022-08-30:
Can you print the value of w、iw、h、ih？

jia0511, 2022-08-31:
@ruiyingy1213 

Thank you very much for your help, now I have found the reason, thank you for your professional support.

closed 2022-08-31 by jia0511