Xilinx/Vitis-AI #964: Error when compiling quantized pytorch model with vai_c_xir: [UNILOG][Check Failed: shift_cut >= 0][/home/xbuild/conda-bld/xcompiler_1606385139459/work/src/slnode/DPUv2/SlNodeDptConv.cpp:55][XCOM_DATA_OUTRANGE][Data value is out of range!]

wgshun, 2022-08-11:
Hello,

I quantized a pytorch model and got an .xmodel. During compilation,  I get the following error:

`* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20220810-215919-99"
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_CUSTOMIZED
[UNILOG][INFO] Graph name: Resnet, with op num: 604
[UNILOG][INFO] Begin to compile...
[UNILOG][Check Failed: shift_cut >= 0][/home/xbuild/conda-bld/xcompiler_1606385139459/work/src/slnode/DPUv2/SlNodeDptConv.cpp:55][XCOM_DATA_OUTRANGE][Data value is out of range!] 
*** Check failure stack trace: ***
Aborted (core dumped)`

Has anyone had the same error or knows what might be causing this error message? 

Hope you can help me.
Thank you!

wgshun, 2022-08-18:
Hi.

niuxjxlnx, 2022-09-05:
Hi, @wgshun 

I noticed there are lots of scale layers converted from BatchNorm layer in your model, this is the root cause leads to deployment problem, which will make CONV layers feature maps not quantization friendly, deployment failure or not-acceptable performance of final deployment.

Suggest moving the BatchNorm layers to the output position of CONV layers, then they will be merged info CONV layers and will get reasonable deployment result.

closed 2022-10-18 by niuxjxlnx