Xilinx/Vitis-AI #965: WARNING when compiling quantized pytorch model with vai_c_xir: MaxPool2d or AvgPool2d has been allocated to CPU.

wgshun, 2022-08-11:
Hello,

I quantized a pytorch model and got an .xmodel. During compilation, I get the following warning:

```
[UNILOG][WARNING] xir::Op{name = Resnet__Resnet_Sequential_body__bottleneck_IR_1__MaxPool2d_shortcut_layer__9252, type = pool-fix} has been allocated to CPU.
[UNILOG][WARNING] xir::Op{name = Resnet__Resnet_Sequential_body__bottleneck_IR_2__AvgPool2d_shortcut_layer__9320, type = pool-fix} has been allocated to CPU.
```

Has anyone had the same error or knows what might be causing this error message?

Hope you can help me.
Thank you!

[image]

wanghong4compiler, 2022-08-11:
Hi @wgshun ,
which hardware target did you choose?

wgshun, 2022-08-12:
@wanghong4compiler 
Hi,
`[UNILOG][INFO] Target architecture: DPUCZDX8G_CUSTOMIZED`

sh39o, 2022-08-12:
Hi @wgshun ,
You can upgrade to vitis-ai >= 1.4, or you can remove the pool manually from your code.
(The pool with kernel size 1 and padding 0 doesn't modify the data, so you can delete the pool safely).

closed 2022-08-14 by wgshun