Xilinx/Vitis-AI #966: Excluding nodes bug in OFA.

LorenzoSun-V, 2022-08-12:
When I use OFAPruner.ofa_model() to exclude some nodes, it just printed the excluded nodes but not actually excluded the nodes.

So I read the OFA source code, and found out that the program didn't exclude the nodes because of missing of "contiune":

[image]

DehuaTang, 2022-08-13:
This is just a hint as to which nodes will be excluded. There are other operations later to exclude these nodes.
'excludes' should be a list of module name, For example： ['model.10.conv', 'model.13.cv1.conv']

LorenzoSun-V, 2022-08-17:
Hi @DehuaTang.

If I remove the 'continue', the 'to_update_dict' in following code will also operate on nodes I want to exclude so the program will hint shape error.
[image]

DehuaTang, 2022-08-17:
The original intent here was to replace all convolutions and BNs, and the code that follows will exclude the modules. 
Here is the correct. 😂

LorenzoSun-V, 2022-08-17:
OK, I'll try OFA later. Thanks for your reply!

DehuaTang, 2022-08-17:
Good Luck !

LorenzoSun-V, 2022-08-18:
Hi @DehuaTang.
I'm trying to use one_step pruner to prune my YOLOX model. The YOLOX model is defined as this released [project](https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/pt_yolox_TT100K_640_640_73G_2.5/model.yaml#L28). (SiLU replaced by LeakyReLU, + / cat replaced by pytorch_nndct.nn.modules.functional.Add() / pytorch_nndct.nn.modules.functional.Cat(),  QuantStub() / DeQuantStub() ... These op replacements are to deploy model.)

When I run my search program, the scores of subnets are very low:
[image]

This must be abnormal. Do you have any ideas about this?

DehuaTang, 2022-08-19:
When doing model compression，‘ + / cat replaced by pytorch_nndct.nn.modules.functional.Add() / pytorch_nndct.nn.modules.functional.Cat(), QuantStub() / DeQuantStub() ... ’ these operations are not required. These operations cause the model to enter the quantization mode of QAT, resulting in very low accuracy.
Try again！

LorenzoSun-V, 2022-08-19:
I've tried to exclude the additional qat ops in the code like this:
[image]

And modify the model.py to turn the ops to torch.cat/+ , ... . I also delete the QuantStub() / DeQuantStub(). But the scores are still very low.

LorenzoSun-V, 2022-08-19:
Maybe you can download this [project](https://drive.google.com/drive/folders/1--SMqxDolYUydsNNH4nQxIdO2CTn9Zc1) to reproduce the problem.

My pruning code is in ./yolox/prune/one_step_xilinx/search.py, the trained weights is ./YOLOX_outputs/yolox_m_coco_float/best_ckpt.pth.

LorenzoSun-V, 2022-08-22:
Hi, I also have some confusion of the whole workflow about float model training -> float model pruning -> float model quantizing.  Would you please refer this [relpy](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1220378330) for the details.