Xilinx/Vitis-AI #969: QAT of YOLOX

LorenzoSun-V, 2022-08-15:
In released [yolox](https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/pt_yolox_TT100K_640_640_73G_2.5/model.yaml), QAT has been used.

I have generated a convert_qat.pth and want to export convert_qat.pth to xmodel.

However, the "qat_convert.sh" script has two undefined cmd params(marked in the red box of the following picture) and really confuse me:
[image]

I delete the params in my script and the program hit me to do calib before export xmodel like PTQ：
[image]

So, how to skip calib step and export xmodel by my convert_qat.pth directly?

yuwang-xlnx, 2022-08-15:
Hi @LorenzoSun-V,

- --nndct_equalization
- --nndct_param_corr

These two cmd parameters are defined in vai_q_pytorch. When you call "import pytorch_nndct", they are introduced into the current command.
Very confusing indeed, I suggest forgetting about this and using the new API.
Please refer to this: https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_qat.py#L631

LorenzoSun-V, 2022-08-15:
Thanks @yuwang-xlnx ! It's really helpful. I'll try the new API later.

LorenzoSun-V, 2022-08-16:
Hi @yuwang-xlnx ,
I've tried the new API and encounter some issues.
I use the following code to export xmodel:
[image]

However, when I use the deployable_model to infer COCO val set, I got:
[image]

I also use exported xmodel to infer COCO val set on VCK5000 and got predicted bboxes and classes, but I got lower results than those on GPU(To evaluate the xmodel, I use VCK5000 to obtain the result json file and use pycocotools to calculate APs):
[image]

So, is it normal to have a 4% loss of AP:IoU=0.50:0.95?

yuwang-xlnx, 2022-08-16:
Hi @LorenzoSun-V,

Is the AP of deployable model consistent with that tested during training? If it is the same, then it is likely that the preprocessing on VCK5000 is different from that on GPU.

LorenzoSun-V, 2022-08-17:
Thanks @yuwang-xlnx . I modify the conf_threshold in prototxt from 0.25(used in demo inference on GPU) to 0.01(used in AP calculation on GPU) and the APs on VCK5000 are equal to those on GPU.
[image]

The same results on VCK5000:
[image]

LorenzoSun-V, 2022-08-25:
Hi @yuwang-xlnx,

I'm working on deploying the pruning YOLOX model. Since the pruning methods and QAT are fully decoupled(refer this [reply](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1221736723)), I retrained another YOLOX model which only including torch original operations.

However, when I use the retrained original pth to QAT, I encountered the shape error in this stem operation. 

The following picture shows the difference between original and QAT stem operations.
[image]

The QAT stem operation just includes a conv layer but not splitting channels and concrating. Why don't the QAT stem operation be the same to the original one? Is it because the DPU does not support this kind of operation(split channels)?

yuwang-xlnx, 2022-08-25:
Hi @LorenzoSun-V,
Sorry I didn't understand what you meant. How did you get the pruned slim model?

LorenzoSun-V, 2022-08-25:
Sorry, I may not have expressed clearly.

In previous QA, I trained the float model based on modified model.py(following QAT rules, details are in this [reply](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1220378330)) and one operation named 'stem' of this modified model.py is class 'FocusDeploy':
[image]

However, an error occurred when I prune the model. This [reply](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1221736723) said "Pruning and QAT are fully decoupled." So I retrained YOLOX model based on an original model.py(without QAT operations like pytorch_nndct.nn.modules.functional.Cat/Add, including the 'FocusDeploy'), the class 'FocusDeploy' in original model.py is class 'Focus':
[image]

When I used this new retrained model to QAT, I encountered the shape error. This error is due to the difference between 'FocusDeploy' and 'Focus' obviously.
Assumed a input tensor with shape of [16,3,640,640], the original 'Focus' splits the tensor to 4 tensors with shape of [16, 3, 320, 320] and then concatenates the 4 tensors to a single tensor with shape of [16,12,320,320]. The the [16,12,320,320] tensor is as the input of the conv layer. But the 'FocusDeploy' takes the tensor with shape of [16,3,640,640] as a input directly. 

So my question "Why don't the QAT stem operation be the same to the original one? Is it because the DPU does not support this kind of operation(split channels)?" may translate to "Why is the FocusDeploy in QAT not the same as the Foucs in original model?"

yuwang-xlnx, 2022-08-26:
Hi @LorenzoSun-V,

The stem operation should be the same with the original one. 
When you do QAT, what you need to do is to wrap the function operation to a `torch.nn.Module` object.
The rewriting process  does not change the model's forward and the new network is fully equivalent to the original network.

LorenzoSun-V, 2022-08-26:
OK, but I'm not sure that slice operation is supported by Vitis-AI and VCK5000.
The code in the green box of the following picture shows slice opertation.
[image]

LorenzoSun-V, 2022-08-26:
I've used a toy model only containing the stem operation to export xmodel successfully.
The xmodel is:
[image]

I think it's ok to deploy the original model. Thanks.

closed 2022-08-26 by LorenzoSun-V

aiqiangbrave, 2022-11-18:
hi, sorry，please ask the following path questions:

Does the first path refer to the .pth file generated by the quant_PTQ.sh script?

What about the second and third paths??
Thank you~~~
Thank you~~~
[image]

aiqiangbrave, 2022-11-19:
> Hi @LorenzoSun-V,
> 
> The stem operation should be the same with the original one. When you do QAT, what you need to do is to wrap the function operation to a `torch.nn.Module` object. The rewriting process does not change the model's forward and the new network is fully equivalent to the original network.
> 
> The class `Focus` and `FocusDeploy` are obviously not the same, so make sure they are still equivalent after wrapping the `torch.cat`.
[Could you help me analyze the size of the model? ](https://github.com/Xilinx/Vitis-AI/issues/969#issuecomment-1214846043)I remember that the xmodel model after yolov4 compression is about 7M, but the model after yolox compression is only twice shrunk (about 90M), is this normal? Thank you so much

aiqiangbrave, 2022-11-19:
> Hi @LorenzoSun-V,
>
> The stem operation should be the same with the original one. When you do QAT, what you need to do is to wrap the function operation to a `torch.nn.Module` object. The rewriting [...]

### **[Could you help me analyze the size of the model? I remember that the xmodel model after yolov4 compression is about 7M, but the model after yolox compression is only twice shrunk (about 90M), is this normal? Thank you so much**

LorenzoSun-V, 2022-11-21:
@aiqiangbrave , it's normal. I also got a YOLOX xmodel of 90M.

aiqiangbrave, 2022-11-21:
Thank you very much，Good morning~~~（China time）
> @aiqiangbrave , it's normal. I also got a YOLOX xmodel of 90M.
I used the officially quantified xmodel file to compile, but the error was displayed. Have you ever encountered it? Thank you

[image]

LorenzoSun-V, 2022-11-21:
No, I didn't encounter this error before.
I used `vai_c_xir -x your_xmodel_file.xmodel -a /your/arch_vck_file.json -o /your_output_dir -n your_output_xmodel_name` to complie in docker container.

aiqiangbrave, 2022-11-21:
> No, I didn't encounter this error before. I used `vai_c_xir -x your_xmodel_file.xmodel -a /your/arch_vck_file.json -o /your_output_dir -n your_output_xmodel_name` to complie in docker container.

yes,yes,yes,I also use this format.
Can I ask you two path questions? Maybe I got the weight path wrong. Thank you very much~~~
[image]
What exactly do these two weights refer to?

LorenzoSun-V, 2022-11-21:
I didn't use the official script to generate xmodel.
Maybe you can refer this [answer](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1225366862).

aiqiangbrave, 2022-11-21:
> 

Ok ,thank you

aiqiangbrave, 2022-11-21:
> 

hi,Can you refer to your quantified script? (If convenient)

aiqiangbrave, 2022-11-21:
> I didn't use the official script to generate xmodel. Maybe you can refer this [answer](https://github.com/Xilinx/Vitis-AI/issues/912#issuecomment-1225366862).

hi,Can I  refer to your quantified script? (If convenient)

LorenzoSun-V, 2022-11-24:
Actually, I just used `python tools/train/py -f {your qat config file} -d 4 -b 32` to train the model. And then use `convert_qat.py` to convert `model.pth` to `depolyable.pth` which used for generating xmodel.

In class ExpDeployQat, the function get_model() is like:
```
    def get_model(self):
        from yolox.models.yolox_deploy import YOLOX
        from yolox.models.yolo_pafpn_deploy import YOLOPAFPN
        from yolox.models.yolo_head_deploy import YOLOXHead

        def init_yolo(M):
            for m in M.modules():
                if isinstance(m, nn.BatchNorm2d):
                    m.eps = 1e-3
                    m.momentum = 0.03

        if getattr(self, "model", None) is None:
            in_channels = [256, 512, 1024]
            backbone = YOLOPAFPN(self.depth, self.width, in_channels=in_channels, act=self.act)
            head = YOLOXHead(self.num_classes, self.width, in_channels=in_channels, act=self.act)
            self.model = YOLOX(backbone, head)

        self.model.apply(init_yolo)
        self.model.head.initialize_biases(1e-2)

        if self.is_qat:
            if self.float_ckpt:
                ckpt = torch.load(self.float_ckpt)
                print('Loading float model weight for QAT from {}'.format(self.float_ckpt))
                self.model.load_state_dict(ckpt["model"])
            dummy_input = torch.randn([1, 3, self.input_size[0], self.input_size[1]], dtype=torch.float32).cuda()
            logger.info(f"dummy_input size: {self.input_size[0]} x {self.input_size[1]}")
            self.qat_processor = QatProcessor(self.model, dummy_input, bitwidth=8, mix_bit=False)
            self.model = self.qat_processor.trainable_model()

        return self.model
```

In addition, function convert_qat is like:
```
    def convert_qat(self, qat_model, ckpt_path, cvt_dir):
        if not os.path.exists(cvt_dir):
            os.makedirs(cvt_dir)
        print("loading checkpoint from {}".format(ckpt_path))
        ckpt = torch.load(ckpt_path) 
        qat_model.load_state_dict(ckpt["model"])
        deployable_net = self.qat_processor.to_deployable(qat_model, output_dir=cvt_dir)
        print("Save converted QAT results in {}".format(cvt_dir))
        return deployable_net
```

Hope it helps you.

Shreyas-NR, 2022-11-25:
Hi @LorenzoSun-V 
I'm also working on YOLOX with the custom dataset. 
I trained the float model and quantized it.
Now I want to use the QAT scripts, could you help me with the steps to follow?

Also, Do you have a reference script for the application code to run on the board? I'm confused about how to post-process the 3 outputs of the cat() from the quantized model.

LorenzoSun-V, 2022-11-25:
Hi @Shreyas-NR 

About QAT(take COCO as an example):
First of all, I used 
```
export CUDA_VISIBLE_DEVICES=1,2,3,4
GPU_NUM=4
BATCH_SIZE=32
CFG=exps/example/custom/yolox_m_coco_qat.py
python tools/train.py -f ${CFG} -d ${GPU_NUM} -b ${BATCH_SIZE}
```
to train QAT model.

Then, I used
```
import argparse
import sys
import torch
sys.path.append('.')
from yolox.exp import get_exp
from yolox.evaluators import COCOEvaluator
from yolox.data import COCODataset, ValTransform

def make_parser():
    parser = argparse.ArgumentParser("YOLOX Eval")
    parser.add_argument("-expn", "--experiment-name", type=str, default=None)
    parser.add_argument("-n", "--name", type=str, default=None, help="model name")
    parser.add_argument("-b", "--batch-size", type=int, default=128, help="batch size")
    parser.add_argument(
        "-f",
        "--exp_file",
        default=None,
        type=str,
        help="pls input your expriment description file",
    )
    parser.add_argument("-c", "--ckpt", default=None, type=str, help="ckpt for eval")
    parser.add_argument("--cvt_dir", default=None, type=str, help="directory for the converted results")
    return parser

def main(args):
    exp = get_exp(args.exp_file, args.name)
    if not args.experiment_name:
        args.experiment_name = exp.exp_name
    model = exp.get_model()
    evaluator = exp.get_evaluator(args.batch_size, False, False, False, quantized=True)
    deployable_model = exp.convert_qat(model, args.ckpt, args.cvt_dir)
    print(deployable_model)
    *_, summary = evaluator.evaluate(deployable_model, model, False, False, None, None, exp.test_size, False)
    print("\n" + summary)

    deployable_model = exp.qat_processor.deployable_model(args.cvt_dir, used_for_xmodel=True)
    valdataset = COCODataset(
        data_dir=exp.data_dir,
        json_file=exp.val_ann,
        name="val2017",
        img_size=exp.test_size,
        preproc=ValTransform(legacy=False),
    )
    val_subset = torch.utils.data.Subset(valdataset, list(range(1)))
    subset_loader = torch.utils.data.DataLoader(val_subset, batch_size=1, shuffle=False, num_workers=4, pin_memory=True)
    for imgs, _, _, _ in subset_loader:
        deployable_model(imgs)
    exp.qat_processor.export_xmodel(args.cvt_dir)

if __name__ == "__main__":
    args = make_parser().parse_args()
    main(args)

```
to generate and infer `deployable.pth` on COCO.
[... 167 lines cut ...]

    dist_url = "auto" if args.dist_url is None else args.dist_url
    launch(
        main,
        num_gpu,
        args.num_machines,
        args.machine_rank,
        backend=args.dist_backend,
        dist_url=dist_url,
        args=(exp, args),
    )
```
 convert `deployable.pth`  to `YOLOX_int.xmodel`.

The config file `exps/example/custom/yolox_m_coco_qat.py` mentioned above defines `YOLOX model` following QAT rules( Information about Xilinx QAT rules is [here](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/vai_q_pytorch-QAT) ). 

In a word, QAT steps are:
1. rewrite model.py following QAT model rules
2. retrain QAT model 
3. generate `deployable.pth`
4. convert `deployable.pth` to `xmode`

About post-porcess:
If you wanna depoly yolox xmodel on the borad(infer a jpeg image on the board), you are supposed to refer [this code](https://github.com/Xilinx/Vitis-AI/blob/master/examples/Vitis-AI-Library/samples/yolovx/test_jpeg_yolovx.cpp).
About post-process,  you can refer [here](https://github.com/Xilinx/Vitis-AI/blob/06a72491bd9fc72bbc743ea1753d005cbeb6acc8/src/Vitis-AI-Library/yolovx/src/yolovx_imp.cpp#L80). For futher code, you can read VitisAi source cc code.

Shreyas-NR, 2022-11-25:
Thank you so much @LorenzoSun-V for your response. I'll get back if I get stuck anywhere.

LorenzoSun-V, 2022-11-25:
You are welcome @Shreyas-NR.
YOLOX README.md provided by Xilinx is so outdated that it confused me for a long time.
You'd better follow the up-to-date document like `ug1414-vitis-ai-v2.5` when coding.