Xilinx/Vitis-AI #970: quant for SegNet encount problem

DeepTrial, 2022-08-15:
Vitis-ai tool has generate the converted segnet mode(attachment)l, however the following error jump out:

`File "optimizor.py", line 143, in <module>
    quantization(title=title, model_name=model_name)
  File "optimizor.py", line 110, in quantization
    mean_iou = evaluate(quant_model, 224)
  File "/workspace/Newrizon/evalution/Segmentation-Pytorch/load_model.py", line 92, in evaluate
    save_result=True)
  File "/workspace/Newrizon/evalution/Segmentation-Pytorch/builders/validation_builder.py", line 214, in predict_multiscale_sliding
    tile_output = model(tile_image)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "quantize_result/SegNet.py", line 89, in forward
    output_module_7_0,output_module_7_1 = self.module_7(output_module_0)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/maxpool.py", line 40, in forward
    output = quantize_tensors([output], self.node)[0]
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/quantization/utils.py", line 91, in quantize_tensors
    tensor_type))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 115, in do_scan
    if isinstance(res.values, torch.Tensor):
AttributeError: 'tuple' object has no attribute 'values'`

[SegNet.zip](https://github.com/Xilinx/Vitis-AI/files/9335816/SegNet.zip)

kewang-xlnx, 2022-09-01:
Hi @DeepTrial. Quantization of  Maxpool2d operator with parameter "return_indices=True" is not supported in recent vitis-ai tool. It will be supported in Vitis-AI 3.0.

HuiJu1218, 2022-09-19:
Hi,
when I try to quantize model (yolov4-tiny) in vitis-ai 2.5 pytorch container, I got same error as following. This error occured when I using `model.eval()`. My model  was reference from [this github](https://github.com/bubbliiiing/yolov4-tiny-pytorch). Does anyone has idea about this?
```
Traceback (most recent call last):
  File "quantize.py", line 174, in <module>
    run_main()
  File "quantize.py", line 167, in run_main
    quantize(args.build_dir,args.quant_mode,args.batchsize)
  File "quantize.py", line 134, in quantize
    test(quantized_model, device, test_loader)
  File "/workspace/test_vitis-ai-tutorials-09/09-mnist_pyt/files/common.py", line 88, in test
    outputs         = model(images)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "build/quant_model/YoloBody.py", line 72, in forward
    output_module_7_0,output_module_7_1 = self.module_7(input=output_module_0, split_size=32, dim=1)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/nn/modules/module_template.py", line 165, in forward
    output = quantize_tensors([output], self.node)[0]
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/quantization/utils.py", line 91, in quantize_tensors
    tensor_type))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 115, in do_scan
    if isinstance(res.values, torch.Tensor):
AttributeError: 'tuple' object has no attribute 'values'
```

HuiJu1218, 2022-09-23:
Hi,
I have solve my problem. I found vitis-ai not support .split and .chunk from #699 and #685, so I have use tensor slicing to slic the tensor. Thanks.

closed 2022-12-21 by DeepTrial