Xilinx/Vitis-AI #972: vai_c_tensorflow2 error: [UNILOG][FATAL][XCOM_OPERATION_FAILED][The supposed operation is failed!]

jschneider66, 2022-08-15:
Hello, 
   I am using the Vitis AI 2.5 with the latest Aug 2022 patches for the following:
            target_factory-2.5.0-py36h680af44_202.tar.bz2
            target_factory-2.5.0-py37h680af44_202.tar.bz2
            vart-2.5.0-py36h07a2524_202.tar.bz2
            vart-2.5.0-py37h07a2524_202.tar.bz2
            xcompiler-2.5.0-py36hea4fdf2_202.tar.bz2
            xcompiler-2.5.0-py37hea4fdf2_202.tar.bz2
            xir-2.5.0-py36h893bffd_202.tar.bz2
            xir-2.5.0-py37h893bffd_202.tar.bz2
            xnnc-2.5.0-py37hd51751d_201.tar.bz2
   
   I was able to successfully quantize my model but I get the following compile error.    Any suggestions?

 vai_c_tensorflow2 \
            --model           build/quant_model/q_model.h5 \
            --arch                /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json \
            --output_dir      build/compiled \
            --net_name        customcnn

(vitis-ai-tensorflow2) Vitis-AI /workspace> 
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['build/quant_model/q_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/customcnn_DPUCZDX8G_ISA1_B4096_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/raidio_nn/build/quant_model/q_model.h5
[INFO] keras version: 2.8.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|██████████| 11/11 [00:00<00:00, 18641.35it/s]
[INFO] infer shape (NHWC)  :100%|██████████| 18/18 [00:00<00:00, 6626.07it/s]
[INFO] perform level-0 opt :100%|██████████| 2/2 [00:00<00:00, 1020.76it/s]
[INFO] perform level-1 opt :100%|██████████| 2/2 [00:00<00:00, 3908.95it/s]
[INFO] generate xmodel     :100%|██████████| 18/18 [00:00<00:00, 439.62it/s]
[INFO] dump xmodel: /tmp/customcnn_DPUCZDX8G_ISA1_B4096_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
[UNILOG][INFO] Graph name: model, with op num: 26
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = quant_conv2d, type = conv2d-fix} has been assigned to CPU: ["kernel"(512x2) is not in DPU supported range {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16}].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_weights_new, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_bias_new, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
**[UNILOG][FATAL][XCOM_OPERATION_FAILED][The supposed operation is failed!]
*** Check failure stack trace: *****
Aborted

wanghong4compiler, 2022-08-17:
Hi @jschneider66,
could you help share the xmodel file /tmp/customcnn_DPUCZDX8G_ISA1_B4096_org.xmodel, so that I can reproduce this issue?

jschneider66, 2022-08-24:
Hi @wanghong4compiler   I cannot post the model file yet.  But, I determined the failing layer is a max_pooling2d layer.   The tensor shape into the layer is {1,2,2048,30} and the output of that layer is supposed to be {1,1,128,30}.    The layer before that is a conv2d layer.

jschneider66, 2022-08-26:
The issue was found and solved, posting the solution for others.
On the MaxPooling2D layer our pooling size and strides were set to  (16,16) but the tensor size coming our the previous layer was (2,2048,30).   This causes the compiler to crash, the solution was to fix the pooling_size and strides on that layer to be (2,16) to match the tensor size.  
Also, found that Vitis AI Inspector incorrectly states that MaxPooling2D is not supported when in fact it is supported.

closed 2022-08-26 by jschneider66