Xilinx/Vitis-AI #973: Modify DPU type (3136, 4096 and so on) for KV260

vaan2010, 2022-08-17:
Hi,
I want to use KV260 for some custom applications, and I find various DPU types in KV260 such like 1024, 3136, 4096 and so on.
The models from Vitis AI github are compiled with DPU 4096, But my KV260 is DPU 3136.
So I want to modify the DPU type from 3136 to 4096 through Vivado, and another problem show up: KV260 has no DPU IP in Vivado!

Does anyone could give me some suggestions or Xilinx tutorial for build different DPU type for KV260?

Thanks,
Norris

jiaz-xlnx, 2022-08-22:
Hi @vaan2010 How do you modify the dpu settting?  the DPU IP is in the project ( kv260/overlays/dpu_ip). you can check it. the config file is in kv260/overlays/examples/aibox-reid/dpu_conf.vh. you can modify this file baseing on your related case to configue DPU settings.

vaan2010, 2022-08-22:
Hi @jiaz-xlnx ,
I found the setting of DPU type in dpu_conf.vh, but I want to know why kria series have no DPU ip in Vivado?
The files at overlays/dpu_ip are from Vitis not Vivado, that's mean we only can modify type of DPU in Vitis?

Looking forward your reply and thanks.

Norris

jiaz-xlnx, 2022-08-22:
@vaan2010  The kv260 case only supports developping in Vitis flow now.  The Vivado tool doesn't integrate DPU IP internally. you can check this link. https://github.com/Xilinx/Vitis-AI/tree/master/reference_design . The dpu can run in vivado flow for zcu102 board.

vaan2010, 2022-08-23:
@jiaz-xlnx OK I see. Thanks for your reply. I still have some questions:
1. So Xilinx will integrate DPU IP for Kria Series in Vivado at next Vivado version update?
2. So the block design of DPU in Kris Series can't be viewable currently, I can only modify the parameter in dpu_conf.vh but can't check the connection of DPU with other IPs?

Looking forward your reply and thanks.

Norris

jiaz-xlnx, 2022-08-23:
@vaan2010 
1. we don't have plan to integrate DPU IP in next Vivado version.
2. you can check the  connection of the block design. there is a vivado project in the case folder. you can search the *.xpr file. yes. it only supports modify the parameters using the dpu_conf.vh file now.

closed 2022-09-06 by qianglin-xlnx