Xilinx/Vitis-AI #974: (vai-c-xir) Tensor dimensions change when compiling

AdrFebles, 2022-08-17:
Hello!
I've trained a model in Pytorch with tensor dimensions: (1,1,4,6), where the order is: (Batch, Channels, Height, Width). When I quantize the model I generate a random input with these dimensions and define the quantizer as follows:
```
input = torch.rand(1, 1, 4, 6)
quantizer = torch_quantizer(quant_mode, model, input,device=device)
```
When I test the quantized model there is no porblem with it, but when I compile the model and print the subraph.png file I realize the order of the tensors is changed and this model wait for an input with this size (1,4,6,1) with order: (Batch, Height, Width, Channels), which produce an abnormal performance of the model when I execute it on the board ZCU102.

[image]

Could you please, help me to understand this behavior?    
Thanks

AdrFebles, 2022-08-22:
Hi!, I have seen in the user guide that DPU works with BHWC format. but I trained the network with BCHW tensor order, is this wrong?

MichaelX99, 2022-08-30:
Every model I've tried to quantize has complained during the [verify_xmodel](https://github.com/Xilinx/Vitis-AI/blob/c80e859e3194146baf84220601f37b7403f1c951/src/Vitis-AI-Quantizer/vai_q_pytorch/nndct_shared/compile/xir_compiler.py#L181) method performed at the end that the pytorch_nndct model output shape is not the same as the XIR output shape.

I even tried the example resnet18_quant files and the error persists. 

THIS IS A HUGE BUG since PyTorch is ONLY B, C, H, W and XIR is ONLY B, H, W, C!!!!!!!!!!!

MichaelX99, 2022-08-30:
I did a bit of digging and this does not occur during 1.4.1.978 git hash 9f3d6db. It seems to be something in the nndct_shared deploy_optimizer. In the 2.x many [optimizations](https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/qproc/utils.py#L294) are called over the nndct_graph and specifically there is a function called [layout_transform](https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/nndct_shared/compile/deploy_optimizer.py#L214) which does NOT change the format.

This is in stark contrast to the 1.4.1 version of [get_deploy_graph_list](https://github.com/Xilinx/Vitis-AI/blob/1.4/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/qproc/utils.py#L258) which calls completely different graph optimizations

michael-person, 2022-08-31:
So the solution to my problem was running data through the quantized_model during the xmodel deployment stage as well as the calibration phase... Some documentation and less finicky tools would be a great addition...

@AdrFebles are you reshaping the data to be B, H, W, C before you pass them to the VART code executed in runCNNautoenc? Or are you keeping it in the PyTorch native B, C, H, W? I believe that you will need to do everything during the quantization and compilation stages using data in B, C, H, W but then during deployment with the final xmodel use B, H, W, C

AdrFebles, 2022-08-31:
> So the solution to my problem was running data through the quantized_model during the xmodel deployment stage as well as the calibration phase... Some documentation and less finicky tools would be a [...]

Hi! @michael-person, yes I followed the steps of this tutorial which helped me understand how to deploy the model on the board: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/11-tf2_var_autoenc
They train and quantize the model in (B,C,H,W) format, but in the running app it's necessary to make a reshape with the input dimensions of the dpu runner.

michael-person, 2022-08-31:
Hmm I'm wondering if maybe the increase in loss you're seeing moving from the float version to the quantized version has to do with an output shape/data type issue rather than a data layout issue. Can you verify that the shapes between `out` and `img` are the same? Also you may need to scale `out` by the `output_scale` value, you can see how its used in the `CPUCalcSoftmax` method [here](https://github.com/Xilinx/Vitis-AI/blob/master/examples/VART/resnet50_mt_py/resnet50.py#L41)

AdrFebles, 2022-09-30:
Thank you @michael-person

qianglin-xlnx, 2023-01-12:
Closing since no activity for more than 3 months, please open a new issue if you still have any questions, thanks.

closed 2023-01-12 by qianglin-xlnx