Xilinx/Vitis-AI #975: Model and DPU fingerprint check fail - PyTorch flow for Vitis-AI

Werterror, 2022-08-17:
Hi all,
I am following Vitis-AI tutorial "PyTorch flow for Vitis-AI":
https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/09-mnist_pyt/README.md

I got through everything without a problem to the step where I try to launch model on zcu102, with commend:
`python3 app_mt.py -m CNN_zcu102.xmodel`
Then after pre-processing of images, when DPU is in use, error stating that there is a miss match in model_fingerprint and dpu_fingerprint:  0x101000016010407 for model and 0x1000020f6014407 for dpu.

Tutorial that I am working on was written for version 1.4 of Vitis-AI while I belive I have image for 2.0.
If my understanding is correct and the problem is miss matching versions of DPU, how do I change the models DPU version to match one that I have on my ZCU102?

Screenshot of an error:
[image]

lishixlnx, 2022-08-23:
try 
env XLNX_ENABLE_FINGERPRINT_CHECK=0   python3 app_mt.py -m CNN_zcu102.xmodel
to check if it works

Werterror, 2022-08-23:
@lishixlnx thank you for replay,
but sadly it didnt worked either. Probably either of DPUs have to be changed so that they match each other.
Image with error:
[image]

lishixlnx, 2022-08-24:
compile your model with the correct fingerprint.
find  arch.json  in  your docker , /opt/vitis_ai/compiler/arch
change the content something like 
   { "target": "DPUCZDX8G_ISA1_B4096"}
to 
  {"fingerprint":"0x0101000016010407"}
where 0x0101000016010407 should be your own dpu_fingerprint in above error message.

Werterror, 2022-08-24:
Hi @lishixlnx,
Again thank you for your reply.
After changing target to fingerprint it still doesnt work,
first error occurs (DPU Fingerprint mismatch), also just to make sure I tried that twice.
The thing is that "meta.json" file that I am editing is created After compiling the model,
it might be more of a description file rather than setting file.

Whole project compressed: https://filetransfer.io/data-package/65crYaP8#link

lishixlnx, 2022-08-28:
please notice, I mean  arch.json , not meta.json.
please compile your model in the docker envrionment.

Werterror, 2022-08-30:
Hi, @lishixlnx, I tried to look for "arch.json" but I can't find any file with such name in the project, that's why I included the files.
According to what you wrote I should edit that file before compilation of the model, that means right after quantization.
But after quantization only json file that I can find in the project is quant_info.json.
Also where else could I compile project if not in the docker?

lishixlnx, 2022-08-31:
in document https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Compiling-for-DPU
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/VAI_C-Usage,
it only mentioned the way of compiling in docker and use the arch.json.
Please follow that standard way.

Werterror, 2022-09-07:
Hi @qianglin-xlnx,
The thing is that arch.json didn't get created during Vits-AI flow hence there is nothing to be edited.
Also project that is subject of this discussion doesn't use arch.json file to compile its model.
I will see if I can figure out how to create one and use it with example project.
BR

qianglin-xlnx, 2022-09-08:
Hi @qianglin-xlnx 
You can create the arch.json. It's one of the inputs for the compiler, as shown in the compile.sh script below.
https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/09-mnist_pyt/files/compile.sh#L20

Werterror, 2022-09-08:
Hi @qianglin-xlnx,
I will try that solution next week and let you know about the results.
Thank you for all the resources.