Xilinx/Vitis-AI #977: How to integrate the DPU xo file into my own project

kkenshin1, 2022-08-18:
I want to integrate the DPU IP into my own application, and I just have the DPU xo file download from [here](https://github.com/Xilinx/Vitis-AI/tree/master/reference_design).

I want to know how to get the weight and DPU instruction data from compiled *.xmodel, and how to transfer them from CPU to FPGA?

Is there any documentations can help me? I only find PG367 and UG1414.

I found [this video](https://www.bilibili.com/video/BV1uC4y1h7n7?spm_id_from=333.337.search-card.all.click&vd_source=1294a6e7fca137ebbbfc01d2d04a1915) in bilibili

In the third part of this video, the speaker show how to integrate the DPU IP, JPEG Decoder IP and Image Resize IP together, that is the same as my current requirement. Is the code opensource? Or is there documentation describing how to implement such design based on Vitis?

SuZhan0322, 2022-08-31:
hi @kkenshin1 , I understand your question is how does xmodel run on dpu, if so, you can refer to https://docs.xilinx.com/r/2.0-English/ug1354-xilinx-ai-sdk/ and https:// docs.xilinx.com/r/2.0-English/ug1414-vitis-ai

kkenshin1, 2022-08-31:
hi @SuZhan0322, I mean that I want to integrate DPU *.xo file with other kernels, and use Vitis flow to develop my own project. I think maybe I can config the registers in DPU and transfer instructions from CPU to FPGA by myself. I want to know does Xilinx Vitis AI support this development method? if yes, how can I parse the *.xmodel file to get instruction and weight data by myself?

bozxilinx, 2022-09-05:
Hi kkenshin1
  currently, we only support XO+Vitis flow for DPU. And does not support User-IP format+Vivado.  So, it seems you want to use User-IP+Vivado, right?

kkenshin1, 2022-09-05:
Hi @bozxilinx, I could use XO+Vitis flow for DPU, but is there any documentations about how to use this flow? Can I use OpenCL API or XRT API to manage all my accelerators (including DPU)?

bozxilinx, 2022-09-05:
Hi Kkenshin1
    there should be a README or makefile in different DPU IP tar ball to introduce how to use the flow. PG files do not include this part. which IP do you want to use?

kkenshin1, 2022-09-05:
Hi @bozxilinx, currently I use DPUCAHX8H and DPUCAHX8L, and I want to integrate some user logic IPs with this DPU, such as using DMA IP to transfer DPU instruction and weight data to HBM, and how can I parse .xmodel file to get DPU instruction?

SuZhan0322, 2022-09-05:
Hi @kkenshin1 ,
You can refer to the UG: https://docs.xilinx.com/r/2.0-English/ug1414-vitis-ai. And you can also run the precompiled model of vai2.5 release directly: https://github.com/Xilinx/Vitis-AI/tree/master/examples/Vitis-AI-Library and https://docs.xilinx.com/r/2.0-English/ug1354-xilinx-ai-sdk/Libraries-and-Samples. 
Thanks!

SuZhan0322, 2022-09-29:
hi @kkenshin1 ,
Any update about this issue? Thanks~

kkenshin1, 2022-09-30:
No, thanks, I will close this issue

closed 2022-09-30 by kkenshin1