Xilinx/Vitis-AI #978: Confusion about  pt_OFA-yolo

codersabrinalee, 2022-08-18:
I find that the structure of  OFA-yolo is similar to yolov5. So can I modify the pt_OFA-yolo project to deploy my own  model  trained by yolov5 project ? In addition, I see “GENETARED BY NNDCT, DO NOT EDIT!” in the document named “yolo_base_0.py”,so how to generate the  model  document?

wangxd-xlnx, 2022-08-24:
Hi @codersabrinalee 

We don‘t recommend using OFA-yolo project to deploy your own trained yolov5 model. These .py files such as 'yolo_base_0.py' are generated by AI Quantizer tool (nndct). Manual modification will result in errors.

Due to its license issue, currently we cannot release the yolov5 model.

qianglin-xlnx, 2022-11-02:
Closing since no activity for more than 2 months, please reopen if you still have question, thanks

closed 2022-11-02 by qianglin-xlnx