Xilinx/Vitis-AI #979: DPUCZD8XG xilinx_zcu102_bsp Petalinux-Build Error

dtroiano1, 2022-08-18:
Hello,

I am working to configure a Petalinux image that can trace CPU functions using vaitrace and Vitis Analyzer. 

I have downloaded the latest 2022.1 Petalinux tools as well as the latest DPUCZ8XG dpu IP from the Xilinx website.

After sourcing the settings.sh and configuring the xilinx_zcu102_bsp rootfs to enable vitis-ai-library, vitis-ai-dbg, vitis-ai-dev, and packagegroup-petalinux-self-hosted, I am running into an "unilog-2.5-r0 do_fetch: Failed to fetch url git://github.com/Xilinx/Vitis-AI.git;protocol=https;branch=master" error. Below is the output of the console. Thanks in advance!

[image]

tianfang-fafafa, 2022-08-22:
hi @dtroiano1 , are you using this recipe https://github.com/Xilinx/Vitis-AI/blob/master/setup/petalinux/recipes-vitis-ai/  ?

dtroiano1, 2022-08-22:
Hello @tianfang-fafafa, yes I have confirmed I am using those recipes but am failing with the same error as above. The project is able to build successfully by leaving the vitis-ai-library, vitis-ai-dbg, vitis-ai-dev packages out of the rootfs.

tianfang-fafafa, 2022-08-23:
hi @dtroiano1 ,  I can fetch source of unilog on my machine.  

[image]

Please check these info:

1. SRC of unilog and other vitis ai runtime is  SRCREV = "c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8" as described in https://github.com/Xilinx/Vitis-AI/blob/master/setup/petalinux/recipes-vitis-ai/vitis-ai-library/vitisai.inc
2. Your machine can access to Internet
3. Share your log file of unilog's do_fetch:
    
[image]