Xilinx/Vitis-AI #980: Error about quant facenet model

jzymessi, 2022-08-19:
When I quant facenet model ,  encountered the following problem:

[VAIQ_NOTE]: =>Finsh dumping data.(quantize_result/deploy_check_data_int/Facenet)
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0819 02:40:39.966396  3196 tensor_imp.cpp:36] [UNILOG][FATAL][XIR_MEANINGLESS_VALUE][The value you set for this parameter makes no sense.] The shape of each dim can only >= 0 or == -1, -1 means that this value has not been specified yet. But tensor "Facenet__Facenet_inception_resnet_backbone__BasicConv2d_model__BasicConv2d_conv2d_4a__Conv2d_conv__input_27"'s dims are {1, -3, 36, 192}.
*** Check failure stack trace: ***
Aborted (core dumped)

**this is my code:**
    quantizer = torch_quantizer(args.quant_mode, model, (input_signature), output_dir=args.output_path, device=device)
    quant_model = quantizer.quant_model
    if args.quant_mode == 'calib':
        quantizer.fast_finetune(eval_fn, (quant_model, LFW_loader))
        quantizer.export_quant_config()
    if args.quant_mode == 'test':
        quantizer.load_ft_param()
        quantizer.export_xmodel(deploy_check=True)
        quantizer.export_onnx_model()

How can I solve it ?

jzymessi, 2022-09-19:
Hi xilinx:
This facenet model is my own model，I use vitsi-ai 2.5 for Quantization. (pytorch conda env)
You can repeat it through this link:
link：https://pan.baidu.com/s/1xiznErlDpe52I_o05wRr-A 
passwd：pygk

carrotandjzy, 2022-10-08:
Sorry, the model file has been expired. Can you provide your model again ?

carrotandjzy, 2023-01-12:
Hi @jzymessi 
I think the quantizable model should forward at least once before exporting xmodel.

niuxjxlnx, 2023-01-13:
@jzymessi ,

Please refer to the example at 
`https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py#L268`
This line of code is to do inference of quantizer.quant_model. It must be done in calibration, test stage of quantization and before exporting xmodel.

closed 2023-01-13 by niuxjxlnx