Xilinx/Vitis-AI #981: Error in vai_q_pytorch at testing phase when converting to xmodel

micro83, 2022-08-19:
Hello,

I'm trying to quantize a PyTorch model with Vitis-AI 2.5. The quant_mode == 'calib' phase went good and module.py & quant_info.json files were generated  successfully.
The problem occurred in the quant_mode == 'test' at the last step [0;32m[VAIQ_NOTE]: =>Converting to xmodel ...[0m

The logging data  including the error is as follows:

`No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[0;32m[VAIQ_NOTE]: Loading NNDCT kernels...[0m

-----------------------------------------
PyTorch version :  1.10.1
3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53) 
[GCC 9.4.0]
-----------------------------------------
 Command line options:
--build_dir    :  ./build
--quant_mode   :  test
--batchsize    :  100
-----------------------------------------
No CUDA devices available..selecting CPU

[0;33m[VAIQ_WARN]: CUDA is not available, change device to CPU[0m

[0;32m[VAIQ_NOTE]: Quant config file is empty, use default quant configuration[0m

[0;32m[VAIQ_NOTE]: Quantization test process start up...[0m

[0;32m[VAIQ_NOTE]: =>Quant Module is in 'cpu'.[0m

[0;32m[VAIQ_NOTE]: =>Parsing CNN...[0m

[0;32m[VAIQ_NOTE]: Start to trace model...[0m

[0;32m[VAIQ_NOTE]: Finish tracing.[0m

[0;32m[VAIQ_NOTE]: Processing ops...[0m

                                                  | 0/12 [00:00<?, ?it/s]
                                                  | 0/12 [00:00<?, ?it/s, OpInfo: name = input_0, type = Param]
████▏                                             | 1/12 [00:00<00:00, 1960.87it/s, OpInfo: name = CNN/Conv1d[conv1]/159, type = _convolution]
████████▎                                         | 2/12 [00:00<00:00, 1223.19it/s, OpInfo: name = CNN/160, type = relu]                      
████████████▌                                     | 3/12 [00:00<00:00, 1395.93it/s, OpInfo: name = CNN/MaxPool1d[pool]/input.3, type = max_pool1d]
████████████████▋                                 | 4/12 [00:00<00:00, 1447.43it/s, OpInfo: name = CNN/Conv1d[conv2]/185, type = _convolution]    
████████████████████▊                             | 5/12 [00:00<00:00, 1538.29it/s, OpInfo: name = CNN/186, type = relu]                      
█████████████████████████                         | 6/12 [00:00<00:00, 1712.19it/s, OpInfo: name = CNN/MaxPool1d[pool]/196, type = max_pool1d]
█████████████████████████████▏                    | 7/12 [00:00<00:00, 1858.94it/s, OpInfo: name = CNN/input.5, type = view]                  
█████████████████████████████████▎                | 8/12 [00:00<00:00, 1876.65it/s, OpInfo: name = CNN/Linear[fc1]/201, type = linear]
█████████████████████████████████████▌            | 9/12 [00:00<00:00, 1903.04it/s, OpInfo: name = CNN/input, type = relu]            
█████████████████████████████████████████▋        | 10/12 [00:00<00:00, 2021.55it/s, OpInfo: name = CNN/Linear[fc2]/203, type = linear]
█████████████████████████████████████████████▊    | 11/12 [00:00<00:00, 2091.64it/s, OpInfo: name = return_0, type = Return]           
██████████████████████████████████████████████████| 12/12 [00:00<00:00, 2211.60it/s, OpInfo: name = return_0, type = Return]

[0;32m[VAIQ_NOTE]: =>Doing weights equalization...[0m

[0;32m[VAIQ_NOTE]: =>Quantizable module is generated.(./build/quant_model/CNN.py)[0m

[0;32m[VAIQ_NOTE]: =>Get module with quantization.[0m

Test set: Accuracy: 366/563 (65.01%)

[0;32m[VAIQ_NOTE]: =>Converting to xmodel ...[0m
Traceback (most recent call last):
  File "quantize.py", line 128, in <module>
    run_main()
  File "quantize.py", line 121, in run_main
    quantize(args.build_dir,args.quant_mode,args.batchsize)
  File "quantize.py", line 96, in quantize
    quantizer.export_xmodel(deploy_check=False, output_dir=quant_model)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 127, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 248, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 321, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/utils.py", line 276, in get_deploy_graph_list
    g_optmizer = DevGraphOptimizer(nndct_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/compile/deploy_optimizer.py", line 40, in __init__
    self._dev_graph.clone_from(nndct_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_graph.py", line 105, in clone_from
    self._top_block.clone_from(src_graph.block, local_map, converted_nodes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_block.py", line 47, in clone_from
    self.append_node(self.owning_graph.create_node_from(node, local_map, converted_nodes))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_graph.py", line 132, in create_node_from
    node.clone_from(src_node, local_map)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_node.py", line 119, in clone_from
    self.op.clone_from(src_node.op, local_map)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/nndct_graph/base_operator.py", line 221, in clone_from
    setattr(self, config, new_value)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/parse/torch_op_def.py", line 203, in padding
    self.set_attr(self.AttrName.PAD, [value[0], value[0]])
TypeError: 'int' object is not subscriptable
`
Do anybody have any thoughts on what caused this error and how to overcome it?

carrotandjzy, 2022-10-08:
The conv1d cannot be deployed in vai-2.5

micro83, 2022-10-09:
Is there any plan to support it in future versions?
I believe supporting DL for 1D signals such as sensor signals is important.

carrotandjzy, 2022-10-10:
It will be released in VAI-3.0

closed 2023-01-13 by niuxjxlnx

zyt1024, 2023-08-17:
> The conv1d cannot be deployed in vai-2.5

Hello, the conv1D is not support in vai-2.5, I use the board is U200, my model contains the Conv1D， but the vai3.0 is not support Conv1D, how can I solve this issue?