Xilinx/Vitis-AI #982: Is it possible to use vaitrace on AWS F1 instance?

williamliao28, 2022-08-19:
Hello,

I have tried to run vaitrace with AWS F1 instance and always get the following error

```
ERROR:root:[]
ERROR:root:xmodel info raw info error
```

So I would like to know if there is any way to do profiling with vaitrace on AWS F1 instance because I need more detailed information on how the operations are executed on DPU circuits.

SuZhan0322, 2022-09-01:
Hi @williamliao28 ,
The vaitrace tool is supported on AWS F1 instance, But the xclbin needs an update to run 2.5 models on F1 instance. We are working on it and will release another AMI soon. thanks~

SuZhan0322, 2022-09-29:
Hi @williamliao28,
Please reopen this issue if you have any other questions. thanks!

closed 2022-09-29 by SuZhan0322