Xilinx/Vitis-AI #983: Compile SOLO model for zcu102/zcu104

RapidEdgeAI, 2022-08-22:
Hello,

I tried to execute the SOLO Vitis-AI-Library example on my MPSoC device.
I downloaded the pt_SOLO_coco_640_640_107G_2.5 model and the .prototxt file for zcu102/zcu104 from the vitis ai 2.5 modelzoo and ran the example successfully.
Then I downloaded the SOLO_coco_640_640_107G_2.5 model for GPU and compiled it using vai_c_xir. I sent it to my device and tried to execute the example again. But I get following error message:

`root@ai-modulebigplnx:~/Vitis-AI/examples/Vitis-AI-Library/samples/solo# ./test_jpeg_solo ~/solo_pt.xmodel sample_solo.jpg
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0820 12:45:49.334309   923 solo_imp.cpp:111] cannot found output tensor: 17762
*** Check failure stack trace: ***
Aborted`

I viewed the compiled .xmodel with Netron and I think the reason for this problem is that the compiled model has only 15 fix2float outputs. The precompiled model for zcu102/zcu104 from the vitis ai modelzoo has 20 outputs.

Why is the precompiled SOLO model different from my compiled model? How was the precompiled model compiled? Do I need to set some extra compiler options for vai_c_xir?

qianglin-xlnx, 2022-08-23:
Hi @RapidEdgeAI 
Yes, you need to set the extra compiler options. You can refer to the following: 
`
vai_c_xir -o solo_pt -n solo_pt -x SOLO_0_int.xmodel -a DPUCZDX8G_ISA1_B4096_arch.json --options {"output_ops":"SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17762_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18113_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18483_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19223_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18853_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18852_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18849_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19222_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19219_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18099_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18098_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19237_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18497_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18127_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17776_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18482_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18479_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17748_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17747_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18867_fix"}
`

RapidEdgeAI, 2022-08-23:
Hi @qianglin-xlnx,
thank you very much. This is exactly what I was looking for. But unfortunately, it doesn't compile. vai_c_xir throws an error that the extra compiler options have a syntax error:
```
/workspace > vai_c_xir -o model_zoo/outputs/ -n solo_pt -x model_zoo/downloaded_models/pt_SOLO_coco_640_640_107G_2.5/quantized/SOLO_0_int.xmodel -a model_zoo/arch.json --options {"output_ops":"SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17762_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18113_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18483_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19223_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18853_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18852_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18849_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19222_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19219_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18099_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18098_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19237_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18497_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18127_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17776_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18482_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18479_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17748_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17747_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18867_fix"}
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/bin/vai_c_xir", line 120, in <module>
    compiler = VAI_XIR_Frontend(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/bin/vai_c_xir", line 93, in __init__
    extra_options = eval(args.options)
  File "<string>", line 1
    {output_ops:SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17762_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18113_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18483_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19223_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18853_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18852_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18849_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19222_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19219_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18099_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18098_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__19237_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18497_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18127_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17776_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18482_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18479_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17748_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__17747_fix,SOLO__SOLO_DecoupledSOLOLightHead_bbox_head__18867_fix}
                                                                                                                             ^
SyntaxError: invalid syntax
```
Just a few other questions related to this topic:
Do I only need these extra compile options for the SOLO Vitis-AI-Library example? Or do I also need those options for my own SOLO Vitis-AI-Library and VART applications?  How do I determine if extra compiler options are needed for a model from the Vitis AI modelzoo? 

Best regards!

qianglin-xlnx, 2022-08-23:
Hi @RapidEdgeAI 

Please add " ' " for the options , as shown below.
[image]

For some model structure, such as conv->eltwise, usually, the xcompiler will take eltwise as output. But if you also want conv as output, you have to set the extra compiler options for the specific outputs.

So far as I know, the following 3 models from Xilinx model zoo need extra compiler options:

- solo_pt
- FADNet_0_pt
- FADNet_pruned_0_pt

RapidEdgeAI, 2022-08-23:
Thank you very much. Now the solo example program executes successfully.
Also thanks for the detailed explanation of the extra compiler options for the xcompiler.

closed 2022-08-23 by RapidEdgeAI