Xilinx/Vitis-AI #984: Run Bert demo on vck5000

jzymessi, 2022-08-23:
I run Bert demo on VCK5000 , I encountered the following problem
[image]
But I run Vitis-ai-Library demo is succeed

SuZhan0322, 2022-08-30:
hi @jzymessi ,
Please run the xrt reset command before switching between different xclbins.
"/opt/xilinx/xrt/bin/xbutil reset --device [device info]"

jzymessi, 2022-08-30:
Hi @SuZhan0322 :

   I reboot the vck5000 card,it's works,I run the demo succeed.

   Could you please tell me some questions about Bert demo?

SuZhan0322, 2022-08-31:
Hi @jzymessi ,
Greate! And the reboot and reset operations are all worked.
Do you have any other issue about BERT Demo?

jzymessi, 2022-08-31:
Hi @SuZhan0322 ：

I have some questions about Bert demo.

1. This Demo is executed by self-built Xclbin. If the structure of Bert model is modified, does Xclbin also need to be modified?

2. Is the bert_base.bm contained in the file a running model? If yes, how is the file generated? (Quantization and compilation)

3. Currently, does Vitis AI 2.5 support the quantization compilation and deployment of PyTorch version Bert model?

4.I found model zoo has bert model [https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/tf_bert-base_SQuAD_128_22.34G_2.5/model.yaml](url) , but I can't compile it.

SuZhan0322, 2022-08-31:
Hi @jzymessi ,
1. Yes, if change the model, xclbin also needs to be updated.
2. The quantized and compiled parameter information of the model is saved in bert_base.bm.
3. VAI2.5 supports the tensorflow version bert and the pytorch version bert is planned to be released in vai3.0.
4. The bert model compilation is not supported by VAI2.5 released compiler, and the quantize/compilation information is saved in the bert_base.bm file.

jzymessi, 2022-09-01:
Hi @SuZhan0322 ，

   Thanks for your reploy. I still have some questions.

   1. If the structure of the Bert model changes (for example, the number of channels changes, the model layer changes, but the operator does not change), whether xclbin needs to be modified in this case.

   2. For the bert_base.bm file, can you tell the detailed process of its generation?
   
Thanks.

SuZhan0322, 2022-09-05:
Hi @jzymessi ,
1. Yes, the xclbin needs to be update.  In VAI 2.5 release, bert's ip is not compatible with other bert models. IPs compatible with different bert models will be released in VAI 3.0, so stay tuned.
2. For the bm file, we just integrate the model parameter files into the bm file, and parse them from the bm file in the bert demo. bm is just a file carrier and storage/transmission method.

jzymessi, 2022-09-08:
Hi @SuZhan0322 ：

     At present, I have trained two Bert models with PyTorch and TF, and I want to deploy them on VCK5000 in BM mode. How can I convert the models into BM files?

Thanks！

jzymessi, 2022-09-30:
Hi @SuZhan0322 ：
     Thanks for your reply, I'm really looking forward to VAI3.0.

closed 2022-09-30 by jzymessi