Xilinx/Vitis-AI #985: Does VCK5000 support int16 quantization quantization CNN model？

jia0511, 2022-08-24:
Hello xilinx,

As far as I know, currently xilinx devices do not support fp16, but I noticed that LSTM supports INT16, I wonder if CV also supports quantization in INT16 format, and I saw that Bert Large supports FP32 and INT8 mixed precision quantization in your introduction, Bert base supports INT16 quantization. If I want to quantize the CNN in CV form to INT16, should the quantization code in modelzoo be modified?

jia0511, 2022-08-25:
Hello xilinx,

Does vitis-ai2.5 support int16 quantization? Associated questions are here：https://github.com/Xilinx/Vitis-AI/issues/712

niuxjxlnx, 2022-08-25:
Quantizer in Vitis-AI 2.5 supports int16 quantization.

jia0511, 2022-08-25:
@niuxjxlnx 

Do all CNN weights support in16 quantization? How does the user end perform int16 format model quantization? It is necessary to modify the quantization value of the weight and activation function in the quantization code api.py? Or the activation function of LSTM and RNN only supports int16 quantization？

niuxjxlnx, 2022-08-26:
@jia0511 :

For how to control bits number of quantization, please refer to Vitis-AI 2.5 user guide [ug1414, ](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Configuration-of-Quantization-Strategy).

LSTM/RNN also support any bits number of quantization in tool, but for deployment of acceptable accuracy, only int16 quantization for activation makes sense.

jia0511, 2022-08-29:
@niuxjxlnx

I reopened the question. After the model is quantized to INT16, I have encountered some problems during deployment and need help to answer.

jzymessi, 2022-08-29:
Hi @niuxjxlnx :

  I already quant and compile int16  resnet50 model in Vitis Ai 2.5 . But when I run classification jpeg demo，I found accuracy is error. 
  
  Is there anything else that needs to be changed during deploy?

[image]

niuxjxlnx, 2022-08-29:
@jia0511 , @jzymessi :

In Vitis-AI 2.5, for CNN model, only qunatizer supports int16 quantization, compiler and runtime does not support it.

jzymessi, 2022-08-30:
@niuxjxlnx ：
   Thanks.
   May I ask when compiler and runtime can support it?

jia0511, 2022-09-02:
@niuxjxlnx 

We would like to know whether the vitis-ai3.0 version, which will be released next year, supports the deployment of INT16, or whether it supports fp16 for quantitative deployment on hardware?

niuxjxlnx, 2022-09-05:
> @niuxjxlnx ： Thanks. May I ask when compiler and runtime can support it?

So far, for CNN model, compiler and runtime has no plan to support INT16 deployment.

niuxjxlnx, 2022-09-05:
> @niuxjxlnx
> 
> We would like to know whether the vitis-ai3.0 version, which will be released next year, supports the deployment of INT16, or whether it supports fp16 for quantitative deployment on hardware?

fp16 model to int8 deployment will be supported in VAI 3.0, but no plan of int16 deployment for CNN model.

closed 2022-09-06 by jia0511