Xilinx/Vitis-AI #987: How to change xmodel's input shape

HappyOgreMagi, 2022-08-24:
I want to change the input shape of pt_SESR-S_DIV2K_360_640_7.48G_2.0 to 360*640，but it is always 256*256. I have tried change "dummy_input "in main_qat.py,but it  is not work .
[image]

wangxd-xlnx, 2022-08-31:
Hi @HappyOgreMagi 

If you want to change the input shape of xmodel, you need to change the data you use for dumping xmodel. 
Just change "dummy_input " parameter in QAT scripts won't work. According to logic of tool, the first image of the given data is used by default.

HappyOgreMagi, 2022-08-31:
> Hi @HappyOgreMagi
> 
> If you want to change the input shape of xmodel, you need to change the data you use for dumping xmodel. Just change "dummy_input " parameter in QAT scripts won't work. According to logic of tool, the first image of the given data is used by default.

HappyOgreMagi, 2022-08-31:
thanks，its worked

closed 2022-08-31 by HappyOgreMagi