Xilinx/Vitis-AI #988: Vitis support for Tensorflow 2 Separable Convolution layers

aghitchner, 2022-08-24:
Hello,

My team is interested in deploying an EfficientDet model on a ZCU102 board. We are using the AutoML repository to train our EfficientDet TF2 model. We are using Vitis-AI to quantize and compile the model for deployment to the board. 

We have ran into several issues with getting the AutoML code to quantize. While most have been solved with workarounds, like turning the subclass keras model into a functional keras model, we have ran into the blocker that Vitis-AI does not support Separable Convolution layers for TF2. We would like to get these layers on the DPU and don't want to revert back to using regular convolution layers as this would make the model too large and slow for our needs. 

Is there plan to support Separable Convolution layers in TF2 in the near future?

aghitchner, 2022-09-07:
@renjiangyong thank you! Is the patch just simply a different branch?

renjiangyong, 2023-06-19:
We had fixed this issue in the latest Vitis-3.0, please use the latest docker to test, thank you.