Xilinx/Vitis-AI #991: DPU debug with VART

OxygenFunction, 2022-08-26:
I have generated a quantized inference model and reference result. also DPU xmodel.
However, error was occurred when generating the DPU inference result,.
[image]
I wonder what is this "Check failed. !the_factory_methods.empty()" mean.

qianglin-xlnx, 2022-08-26:
Hi @OxygenFunction 
Which target board do you use?

OxygenFunction, 2022-08-26:
> Hi @OxygenFunction Which target board do you use?

It would be DPUCZDX8G_ISA1_B4096 on zcu102

qianglin-xlnx, 2022-08-31:
Hi @OxygenFunction 
For zcu102, please run the xdputil command on the board, not in the docker on the host.

closed 2022-09-08 by OxygenFunction