Xilinx/Vitis-AI #992: ImportError: cannot import name 'DecentQTransformWithStringInputs'

anirudhitagi, 2022-08-29:
Greetings,

I am using **Ubuntu 22.04 LTS**, on which I have **Docker version 20.10.17, build 100c701**. I have taken the latest **xilinx/vitis-ai-cpu docker image: 2.5.0.1260   (CPU)**.

I am attempting a simple quantization of the [ssd_mobilenetv2](https://github.com/Xilinx/Vitis-AI/tree/master/model_zoo/model-list/tf_ssdmobilenetv2_coco_300_300_3.75G_2.5). I am following the instructions as mentioned in that readme file yet I am unable to quantize the data. I get the following error:

`(vitis-ai-tensorflow) Vitis-AI /workspace/models/downloads/tf_ssdmobilenetv2_coco_300_300_3.75G_2.5/code/quantize > ./quantize.sh
[INFO]Start Quantize ...
[Config] FLOAT_MODEL: ../../float/frozen_inference_graph.pb
[Config] INPUT_HEIGHT: 300
[Config] INPUT_WIDTH: 300
 [WARNING]Config PREPROCESS_TYPE is not set, please set it in config.ini  

[Config] Q_INPUT_NODE: image_tensor
[Config] Q_OUTPUT_NODE: concat,concat_1
[Config] METHOD: 1
[Config] CALIB_IMAGE_DIR: ../../data/coco2014_minival_8059/image
[Config] CALIB_IMAGE_LIST: ../../data/calib_list.txt
[Config] CALIB_BATCH_SIZE: 50
[Config] CALIB_INPUT_FN: input_fn.calib_input
[Config] CALIB_ITER: 20
[Config] QUANTIZE_DIR: ../../quantized
[Config] GPUS: 0
[Config] TEST_SCRIPT: ../test/ssd_detector.py
using ../../quantized as quantize results dir
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 7, in <module>
    from tensorflow.contrib.decent_q.python.decent_q import run_main
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/__init__.py", line 23, in <module>
    from tensorflow.contrib.decent_q.utils import DecentQTransform
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/utils/__init__.py", line 23, in <module>
    from tensorflow.python.pywrap_tensorflow import DecentQTransformWithStringInputs
ImportError: cannot import name 'DecentQTransformWithStringInputs'`

I am not able to resolve that error of **ImportError: cannot import name 'DecentQTransformWithStringInputs'**. Please advise how to resolve this. 

Please let me know if I need to provide any other information.

Thanks and Regards,
Anirudh Itagi

anirudhitagi, 2022-08-29:
some more context:
(```
vitis-ai-tensorflow) Vitis-AI /workspace/models/downloads/tf_ssdmobilenetv2_coco_300_300_3.75G_2.5 > vai_q_tensorflow inspect --input_frozen_graph=float/frozen_inference_graph.pb 
Op types used: 431 Identity, 412 Const, 64 FusedBatchNorm, 55 Conv2D, 47 Relu6, 21 DepthwiseConv2dNative, 12 BiasAdd, 12 Pack, 12 Reshape, 12 Shape, 12 StridedSlice, 10 Add, 2 ConcatV2, 1 Placeholder, 1 Squeeze

Found 1 possible inputs: (name=image_tensor, type=float(1), shape=[?,300,300,3]) 
Found 2 possible outputs: (name=box_encodings, op=Identity) (name=class_predictions_with_background, op=Identity)
```

xieyi4650, 2022-09-13:
Hi @anirudhitagi 
Have you installed TensorFlow after running a docker image?

xieyi4650, 2022-09-13:
Maybe you can check if run command 'pip install -e requirement.txt', and  check if tensorflow in the requirement.txt

anirudhitagi, 2022-09-13:
Hi @xieyi4650 , thank you for your response.
Yes I have installed TensorFlow after running docker image:
Here is the TensorFlow version:
`Vitis-AI /workspace > conda activate vitis-ai-tensorflow
(vitis-ai-tensorflow) Vitis-AI /workspace > pip show tensorflow
Name: tensorflow
Version: 1.15.2
Summary: TensorFlow is an open source machine learning framework for everyone.
Home-page: https://www.tensorflow.org/
Author: Google Inc.
Author-email: packages@tensorflow.org
License: Apache 2.0
Location: /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages
Requires: absl-py, astor, gast, google-pasta, grpcio, keras-applications, keras-preprocessing, numpy, opt-einsum, protobuf, six, tensorboard, tensorflow-estimator, termcolor, wheel, wrapt
`

I had to do this because I was getting the error saying something like tensorflow module not found.

I did run the 'pip install -e requirement.txt'. here is the content of the requirement.txt:

easydict==1.9
matplotlib==3.1.3
numpy==1.17.0
opencv-python==4.2.0.32
protobuf==3.11.3
tensorflow-gpu==1.15.0
tqdm==4.43.0

xieyi4650, 2022-09-14:
the released docker image has been installed TensorFlow build by us, so you do not need to install TensorFlow. If you install TensorFlow, then there will be encounter an error like this.
So can you try to re-start a docker image and run `conda activate vitis-ai-tensorflow` ,  if you need to install requirement.txt then comment TensorFlow in requirement.txt.

sklckdwn, 2023-01-30:
I have the same issue, how did you solve it ?

zhenzhen-AMD, 2023-07-05:
Hi @anirudhitagi and @sklckdwn ,
I have not reproduced the error.
The steps to reproduce the issue are as follows:
1. cd tf_ssdmobilenetv2_coco_300_300_3.75G_2.5
2. vai_q_tensorflow inspect --input_frozen_graph=float/frozen_inference_graph.pb
> Op types used: 431 Identity, 412 Const, 64 FusedBatchNorm, 55 Conv2D, 47 Relu6, 21 DepthwiseConv2dNative, 12 BiasAdd, 12 Pack, 12 Reshape, 12 Shape, 12 StridedSlice, 10 Add, 2 ConcatV2, 1 Placeholder, 1 Squeeze
Found 1 possible inputs: (name=image_tensor, type=float(1), shape=[?,300,300,3]) 
Found 2 possible outputs: (name=box_encodings, op=Identity) (name=class_predictions_with_background, op=Identity) 
3. cd code/quantize
4. bash quantize.sh

I hope this helps! Let me know if you have any further questions.

zhenzhen-AMD, 2023-07-19:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-07-19 by renjiangyong