Xilinx/Vitis-AI #993: vai_q_tensorflow quantize issue

Afef00, 2022-08-29:
I run Vitis AI CPU docker container from host using ./docker_run.sh xilinx/vitis-ai-cpu:1.3.411. I retrained ResNet50 for iris flower classification.  The notebook runs fine, until the  part when I tried to run vai_q_tensorflow quantize. The quantization runs successfully but I got this warning and I don't get what's the problem
```
NFO: Checking Float Graph...
INFO: Float Graph Check Done.
2022-08-29 10:07:39.136938: W tensorflow/contrib/decent_q/utils/quantize_utils.cc:566] [DECENT_WARNING] Convert mean node global_average_pooling2d/Mean to AvgPool + Squeeze
2022-08-29 10:07:39.140287: W tensorflow/contrib/decent_q/utils/quantize_utils.cc:655] [DECENT_WARNING] Scale output of avg_pool node global_average_pooling2d/Mean to simulate DPU.
INFO: Calibrating for 32 iterations...
100% (32 of 32) |########################| Elapsed Time: 0:01:15 Time:  0:01:15
INFO: Calibration Done.
2022-08-29 10:08:55.545578: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv1_conv/Conv2D is -2147483640. It exceed range [-4, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545607: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block1_0_conv/Conv2D is -2147483644. It exceed range [-10, 16], modify quantize pos from 2 to -2147483632
2022-08-29 10:08:55.545612: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block1_1_conv/Conv2D is -2147483645. It exceed range [-9, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545616: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block1_2_conv/Conv2D is -2147483643. It exceed range [-7, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545619: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block1_3_conv/Conv2D is -2147483646. It exceed range [-10, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545622: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block2_1_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545625: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block2_2_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545628: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block2_3_conv/Conv2D is -2147483645. It exceed range [-9, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545631: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block3_1_conv/Conv2D is -2147483643. It exceed range [-7, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545634: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block3_2_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545637: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv2_block3_3_conv/Conv2D is -2147483646. It exceed range [-9, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545640: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block1_0_conv/Conv2D is -2147483644. It exceed range [-8, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545643: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block1_1_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545645: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block1_2_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545648: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block1_3_conv/Conv2D is -2147483646. It exceed range [-10, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545651: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block2_1_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545664: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block2_2_conv/Conv2D is -2147483645. It exceed range [-7, 16], modify quantize pos from 6 to -2147483632
2022-08-29 10:08:55.545668: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block2_3_conv/Conv2D is -2147483647. It exceed range [-10, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545690: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block3_1_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545695: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block3_2_conv/Conv2D is -2147483645. It exceed range [-9, 16], modify quantize pos from 4 to -2147483632
2022-08-29 10:08:55.545698: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block3_3_conv/Conv2D is -2147483647. It exceed range [-10, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545702: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block4_1_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545706: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block4_2_conv/Conv2D is -2147483645. It exceed range [-7, 16], modify quantize pos from 6 to -2147483632
2022-08-29 10:08:55.545710: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv3_block4_3_conv/Conv2D is -2147483646. It exceed range [-9, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545715: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block1_0_conv/Conv2D is -2147483645. It exceed range [-8, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545718: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block1_1_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545722: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block1_2_conv/Conv2D is -2147483643. It exceed range [-6, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545725: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block1_3_conv/Conv2D is -2147483646. It exceed range [-9, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545730: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block2_1_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545733: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block2_2_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545737: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block2_3_conv/Conv2D is -2147483646. It exceed range [-9, 16], modify quantize pos from 5 to -2147483632
2022-08-29 10:08:55.545740: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1681] [DECENT_WARNING] Shift bias of node conv4_block3_1_conv/Conv2D is -2147483644. It exceed range [-7, 16], modify quantize pos from 5 to -2147483632
[... cut ...] 483632
2022-08-29 10:08:55.545857: W tensorflow/contrib/decent_q/utils/graph_quantizer.cc:1687] [DECENT_WARNING] Shift bias of node dense/MatMul is 2147483645. It exceed range [-5, 16], modify quantize pos from 14 to 2147483643

INFO: Generating Deploy Model...
INFO: Deploy Model Generated.
********************* Quantization Summary *********************      
INFO: Output:      
  quantize_eval_model: quantized_iris/quantize_eval_model.pb      
  deploy_model: quantized_iris/deploy_model.pb
```

Afef00, 2022-09-08:
@qianglin-xlnx 
Any suggestions please?

xieyi4650, 2022-09-13:
Hi @Afef00 
There are some constrains from vai-compiler, so we need to adjust some quantize scale.
The vai_qtensorflow is working correctly.

Afef00, 2022-09-13:
The VAI compiler works fine and I get the X.model. The problem was that when I deployed my model on the ZCU104, when I tried to read the output of the convolution layers, I got nan values. When I tried the same steps with another custom model for flower classification, it worked fine.Maybe the problem was with retraining itself or the dataset I used:

```
import numpy as np
import os
import tensorflow as tf
from tensorflow import keras
X = np.load("features.npy")
y = np.load("label.npy")
X = X / 255.0
X= np.array(X, dtype=np.float32)
y= np.array(y, dtype=np.float32)
X_train= X[:338]
Y_train= y[:338]
X_test=X[338:]
Y_test=y[338:]
input = tf.keras.Input(shape=(224,224,3))
efnet = tf.keras.applications.ResNet50(weights='imagenet',
                                            include_top = False,
                                            input_tensor = input)
# Now that we apply global max pooling.
gap = tf.keras.layers.GlobalAveragePooling2D()(efnet.output)

# Finally, we add a classification layer.
output = tf.keras.layers.Dense(3, activation='softmax', use_bias=True)(gap)

# bind all
func_model = tf.keras.Model(efnet.input, output)
func_model.summary()
func_model.compile(optimizer='adam',
             loss="sparse_categorical_crossentropy",
             metrics=['accuracy'])
hist=func_model.fit(X_train, Y_train,validation_data=(X_test,Y_test),epochs=5)

```

xieyi4650, 2022-09-13:
Hi @Afef00 
I think you can debug flow the next step:

1. evaluate float model accuracy
2. evaluate quantized model accuracy
3. check if the pre-processing of the quantized model and the xmode are the same.

Afef00, 2022-09-13:
OK I will check it. T
hanks!

qianglin-xlnx, 2023-06-17:
Closing since no activity for more than half year, please open a new issue if you still have any questions, thanks

closed 2023-06-17 by qianglin-xlnx