Xilinx/Vitis-AI #994: Question about sharing models in Model-Zoo

daniperfer, 2022-08-30:
Hi:

Looking at the wide variety of models that are in the model zoo, a question has come to me:

Could anyone upload a new model to the model zoo?
If I would be able to train and test a certain model for a specific task with a custom dataset, could I share the model through the model zoo?
Is it open for that kind of contributions?
Are there some minimum requirements (source codes, target boards, etc.) for a model to be shared through the model zoo?

wangxd-xlnx, 2022-08-31:
Hi @daniperfer 

Thanks for your message. For open model zoo, we believe it's a good idea. We were considering setting up an open model zoo for user to upload and share their own models.

However, considering that the uploaded model needs to be verified, especially the verification of the entire deployment process, an appropriate standard or minimum requirements need to be developed. 

Open model zoo module may be added in subsequent versions, please stay tuned.

daniperfer, 2022-09-20:
Thanks for your answer.

Just a follow up question: Do you have a estimated data for that future "open" model zoo, or a hint about what type of requirements a model will need to meet in order to be shareable there?
Thanks.

wangxd-xlnx, 2022-11-01:
Since the exact rules have not been determined, all I can think of is that should have a relatively standardized folder structure similar to the currently released model. It is best to provide float models, quantized models, test code, ’readme‘, and even better if there is training code.

closed 2022-11-17 by qianglin-xlnx