Xilinx/Vitis-AI #999: Performance analysis of DPU kernel and python host program

comihai, 2022-09-03:
Hi,
      
I have deployed an application containing a kernel part on the PL side, i.e., DPU for inference on a deep convolutional neural network (DCNN)  model, and a host program on the PS side, i.e., an application written in python that computes some statistics based on DCNN model results. Now, I want to find out the end-to-end performance of my application, i.e., DPU resource utilization [%], execution time [ms], throughput [# images/s], energy consumption [W].
There is some information in the 'Profiling the Model' chapter in the [Vitis AI manual](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Profiling-the-Model), but I was unable to get the metrics mentioned above by running the command `#python3 -m /usr/bin/vaitrace ./host_program.py ./dcnn.xmodel` on the target board and looking in `xclbin.ex.run_summary`.  What extra steps do I need to take to get all these metrics? Thank you.

li900309, 2022-09-05:
hi @comihai  , i think there are two ways to get the metrics, 1. use --txt to get a test table summary;  python3 -m /usr/bin/vaitrace **--txt** ./host_program.py ./dcnn.xmodel 2. Download the Vitis Unified Software Platform from [here ](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vitis.html) and open the run_summary file with Vitis-Analyzer to get graphical results  https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Using-the-Vitis-Analyzer

comihai, 2022-09-08:
Hi @li900309, thank you for your quick response. I tested both ways and I got an DPU summary that contains few metrics, e.g.,  avg time (ms), performance (GOP/s) and Mem. bandwidth (MB/s). How can I get resource utilization [%] and energy consumption[W] on FPGA side?

li900309, 2022-09-19:
hi @comihai , If you use the DPU bitstream built yourself, you can get resource utilization [%] and energy consumption[W] (estimated value) from vivado reports, if you use pre-built DPU bitstream , you can find the data in DPU product guide [PG338](https://docs.xilinx.com/r/en-US/pg338-dpu)  or [PG338](https://docs.xilinx.com/r/en-US/pg338-dpu)

closed 2022-10-14 by qianglin-xlnx