Xilinx/Vitis-AI #1001: Use Xilinx vck5000 dpu xpfm file, to create xclbin not run

dreamella, 2022-09-06:
we donwload  https://github.com/Xilinx/Vitis-AI/tree/master/reference_design DPUCVDX8H_8pe_normal.tar.gz
modify 8pe.mk add the dir of .xpfm
1 PLATFORM = /opt/xilinx/platforms/xilinx_vck5000_gen3x16_xdma_1_202120_1/xilinx_vck5000_gen3x16_xdma_1_202120_1.xpfm
2 make -f 8pe.mk xclbin
3 crete .xclbin 
./work/2021.2/package.xclbin
./work/complete.xclbin
4 create .json file, we do not use which one json,  At the same, the content of  .json not correct.
./work/2021.2/v++_package_package_guidance.json
./work/2021.2/package/aie_control_config.json
./work/2021.2/package/extractedSystemDiagram.json
./work/2021.2/package/packagedSystemDiagram.json
./work/2021.2/package/libadf/cfg/aie.compiler.report.json
./work/2021.2/package/libadf/cfg/aie.control.config.json
./work/2021.2/package/libadf/cfg/aie.partition.id.json
./work/link/int/vplsettings.json
./work/link/int/systemDiagramModelSlrBaseAddress.json
./work/link/int/aie.router.soln.json
./work/link/int/aie.mapped.json
./work/link/int/aie_control_config.json
./work/link/int/systemDiagramModel.json
./work/link/int/kernel_service.json
./work/link/activetask.json
./work/link/sys_link/iprepo/temp/o0/v_info.json
./work/link/vivado/vpl/prj/prj.runs/impl_1/prep_target_output/libadf/cfg/aie.compiler.report.json
./work/link/vivado/vpl/prj/prj.runs/impl_1/prep_target_output/libadf/cfg/aie.control.config.json
./work/link/vivado/vpl/prj/prj.runs/impl_1/prep_target_output/libadf/cfg/aie.partition.id.json
./work/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ip/ulp_ai_engine_0_0/ulp_ai_engine_0_0_power_data.json
./work/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ip/ulp_ai_engine_0_0/ulp_ai_engine_0_0_io_data.json
./work/link/vivado/vpl/.local/hw_platform/xsa.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_cips_0/bd_0/ip/ip_1/bd_d820_cpm_0_0_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_cips_0/bd_0/ip/ip_1/bd_d820_cpm_0_0_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_cips_0/bd_0/ip/ip_0/bd_d820_pspmc_0_0_sysmon_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_cips_0/bd_0/ip/ip_0/bd_d820_pspmc_0_0_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_cips_0/bd_0/ip/ip_0/bd_d820_pspmc_0_0_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_19/ip_0/bd_09c6_MC2_ddrc_0_phy_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_19/ip_0/bd_09c6_MC2_ddrc_0_phy_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_18/ip_0/bd_09c6_MC3_ddrc_0_phy_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_18/ip_0/bd_09c6_MC3_ddrc_0_phy_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_21/ip_0/bd_09c6_MC0_ddrc_0_phy_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_21/ip_0/bd_09c6_MC0_ddrc_0_phy_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_20/ip_0/bd_09c6_MC1_ddrc_0_phy_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/pfm_top/ip/pfm_top_axi_noc_mc_4int_0/bd_0/ip/ip_20/ip_0/bd_09c6_MC1_ddrc_0_phy_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/ulp/ip/ulp_ai_engine_0_0/ulp_ai_engine_0_0_power_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/prj/xilinx_vck5000_gen3x16_xdma_1_202120_1_emu.gen/sources_1/bd/ulp/ip/ulp_ai_engine_0_0/ulp_ai_engine_0_0_io_data.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/xsa.json
./work/link/vivado/vpl/.local/hw_platform/hw_emu/iprepo/input_products/dcps/partition_metadata.json
./work/link/vivado/vpl/.local/hw_platform/ext_metadata.json
./work/link/vivado/vpl/output/resource.json
./work/v++_link_complete_guidance.json

5 md5sum.txt file not create,  Need I use "md5sum" cmd to create a md5sum.txt?

bozxilinx, 2022-09-07:
Hi dreamella
  what do you mean by "not run"? Do you mean the xclbin can not be generated? Or you can not run application with vitis-ai-library?

  why do you need a json file?
thanks

dreamella, 2022-09-14:
@bozxilinx   xclbin created successful,   but  I can not run application with vitis-ai-library.
every dpu has a  arch.json file.

SuZhan0322, 2022-09-29:
Hi @dreamella ,
Do you mean the arch.json file needed when compiling the xmodel? And the json file is not required when running vitis-ai-library.

SuZhan0322, 2022-10-14:
Hi @dreamella ,
Can you share the detailed error message? Thanks!

SuZhan0322, 2023-06-13:
@dreamella , any update about this issue? Thanks!