Xilinx/Vitis-AI #1002: Compiler vai_c_tensorflow error - KeyError: 'shape'

vongracia, 2022-09-06:
Hello,

I'm trying to compile a quantized yolo4 model and the error is always the same:
'    shape = self.tmp_params['shape']
KeyError: 'shape'
'
the fix given by @zhenzhen-AMD does not seem to work this time :(. Any help?

_Originally posted by @renjiangyong in https://github.com/Xilinx/Vitis-AI/issues/764

zhenzhen-AMD, 2022-09-06:
Hi @vongracia ,
Can floating point and quantized models and code be provided? We need these files to analyze the problem.

vongracia, 2022-09-06:
@zhenzhen-AMD  Sure, thanks for replying so fast :). The files are around 200MB so I uploaded them to wetransfer. It would be amazing that you give me a hand. Process I followed, summary:

---My aim is to make inference on a ZCU102 of a yolov4 darknet trained on custom 4k images dataset---

1. I trained a yolov4 using darknet. The training produced two files: **_.weights and _.cfg**
My training images are 4K resolution. The cfg file used in darknet training is:
[yolov4-custom_SQUARED.zip](https://github.com/Xilinx/Vitis-AI/files/9498646/yolov4-custom_SQUARED.zip)

2. I converted **_.weights and _.cfg to keras** and get two files: yolov4.h5 and anchors.txt. Attached here:
Command: 
_python ../keras-YOLOv3-model-set-master/tools/model_converter/convert.py --yolo4_reorder yolov4-custom_SQUARED.cfg yolov4-custom_SQUARED_best.weights yolov4.h5_
**Convert file**: [convert.zip](https://github.com/Xilinx/Vitis-AI/files/9498968/convert.zip)
**Resulting Files**: https://we.tl/t-0Q1HtpEq5X

3. I converted from **keras to tensorflow frozen graph** and get .pb file. 
Command: 
_python ../keras-YOLOv3-model-set-master/tools/model_converter/keras_to_tensorflow.py --input_model yolov4.h5 --output_model=yolov4_frozen.pb_
**KerasToTensorflow**:  [keras_to_tensorflow.zip](https://github.com/Xilinx/Vitis-AI/files/9498974/keras_to_tensorflow.zip)
**Resulting .pb frozen file**: https://wetransfer.com/downloads/3c8169564114355b1d7dfa36f6e4ae1120220906154718/503149e8ed0a1b1f94e02a76fe4020b320220906154719/450472

5. I **quantized** the frozen model and get another .pb file (attached), using the following command:

vai_q_tensorflow quantize --input_frozen_graph yolov4_frozen.pb
			  --input_fn yolov4_graph_input_keras_fn.calib_input
			  --output_dir ../yolov4_quantized
	          --input_nodes image_input
			  --output_nodes conv2d_93/BiasAdd,conv2d_101/BiasAdd,conv2d_109/BiasAdd
			  --input_shapes ?,416,416,3
			  --gpu 0
			  --calib_iter 20

I got the input_nodes and output_nodes using: https://netron.app/
Input function: [yolov4_graph_input_keras_fn.zip](https://github.com/Xilinx/Vitis-AI/files/9498925/yolov4_graph_input_keras_fn.zip)  (CHECK THIS, PLEASE)
**Resulting .pb quantized file**: https://wetransfer.com/downloads/6e804e185671c0b0428c2969d7353eb320220906154847/c12924dd9618333cba1e4ef1cbc4de1a20220906154847/69f894

5. I try to compile the above quantized model:
TARGET=ZCU102
NET_NAME=dpu_yolov4 
ARCH=/opt/vitis_ai/compiler/arch/DPUCZDX8G/${TARGET}/arch.json

_vai_c_tensorflow --frozen_pb quantize_eval_model.pb
                 --arch ${ARCH}
		 --output_dir ../yolov4_compiled/
		 --net_name ${NET_NAME}
		 --options "{'mode':'normal','save_kernel':'', 'input_shape':'1,416,416,3'}"_

**Error**: if not xmodel.infer_shape(Layout.NHWC):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/ir/xmodel.py", line 506, in infer_shape
    ok, error = self.__do_shape_inference(Layout[self.layout])
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/ir/xmodel.py", line 531, in __do_shape_inference
    xnode.infer_shape(layout)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/ir/xnode.py", line 6993, in infer_shape
    shape = self.tmp_params['shape']
KeyError: 'shape'

It would be amazing if you help me deploy and run this model in a ZCU102. Thanks so much for everything :)

vongracia, 2022-09-07:
If you need any other clarification please let me know @zhenzhen-AMD 

Thanks

zhenzhen-AMD, 2022-09-23:
Hi @vongracia ,

Through analysis, the model contains "softplus",
Currently DPU does not support "softplus", so please replace "softplus" with other ops.

[image]

zhenzhen-AMD, 2022-10-11:
> Caution: This message originated from an External Source. Use proper caution when opening attachments, clicking links, or responding.
>
> Thanks @zhenzhen-AMD<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fzhenzhen-AMD&data=05%7C01%7Czhenzhen.ding%40amd.com%7Ceb77e15bac034d5285ba08daa207e833%7C3dd8961fe4884e608e11a82d994e183d%7C0%7C0%7C638000452093320051%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=%2FgiwgJDf46HuRe1sEfocRDCtBf3Rpa2I8HKV5CDmTKI%3D&reserved=0>
>
> Do you mean when training the yolov4 using darknet? You mean replacing it by Relu?
>
> Thanks
>
> -

[AMD Official Use Only - General]

Hi @***@***.***>,

There are two solutions:
1. When training, you can replace 'softplus' with 'relu'. Eventually 'relu' will run on the DPU.
2. If you keep 'softplus', you need to set "add_shape_info=True" in the quantization stage. When compiling 'softplus' will be treated as 'unknown', and eventually it will run on the CPU.

Regards,
Zhenzhen

zhenzhen-AMD, 2022-12-26:
Hi @vongracia ,

Has this issue been resolved? If it's resolved, I'll close this question later. Please reopen the issue if there are new issues. Thanks.

Regards,
Zhenzhen

closed 2023-01-16 by qianglin-xlnx